KiCad/kicad-library #97: Swap CP and CP2 symbol in device.lib

samdolt, 2015-01-10:
[image]
Currently, default symbol for polarized capacitor is the first symbol on the picture.

- This symbol don't have a "plus" sign and can mislead users.
- IEC60617 standard use the last symbol.

If accepted, I can make the change

reportingsjr, 2015-01-11:
I definitely agree that the first one "CP" should be swapped or even removed. I haven't ever seen that symbol for a polarized capacitor except in KiCad. I have definitely seen the second two quite a lot.

nickoe, 2015-01-11:
Indeed the CP one has always been strange to me, but who cares the others are there.

But in regard to these two pin symbols, they are way too big I think, not to forget that resistors and capacitors are not he same size. Resistors are 500 mil, whilst caps and diodes are 400 mil.

I did make some smaller ones that I used on some project, I can try to dig them up as examples.

samdolt, 2015-01-11:
Is someone uses the first version? If not, maybe it would be better to remove it.

And we can add smaller versions of common symbols such as Rsmall, Csmall, CPsmall.

michal777, 2015-01-11:
If I understood correctly rules 3.1, 3.2 and 3.3 minimum length of capacitor symbol is 400 mils !!!
So big symbols are useless :-(

nickoe, 2015-01-11:
Mmm, where do you see that? Maybe I am looing on the wrong wiki page, but I don't see the bullets numbered with the dot seperator.

https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention

michal777, 2015-01-11:
https://github.com/KiCad/kicad-library/blob/master/KiCad_Library_Convention.txt
(3) General Rules for Symbols: 1, 2, 3.

nickoe, 2015-01-11:
Carl, can you comment on this? Could it be time to add an exception for "small" components?

reportingsjr, 2015-01-12:
I will have to say that I have my own library of capacitors, resistors, etc that are far smaller. The default ones are absolutely massive. I would like to hear Carl's opinion on this as well.

CarlPoirier, 2015-01-12:
I too would vote for removing CP.

As for the size of symbols, they indeed can't be smaller than 400mils if we follow the rules, as Michal found out. Maybe we could specify that some rule applies only to black-box components, and have another rule for discrete components?

michal777, 2015-01-12:
I totally agree. The rules 3.1 and 3.2 are good for most of integred circuits, but for resistors, capacitors, transistors etc 50 mil grid is much better.

nickoe, 2015-01-12:
Indeed, the question should then be what size. One should also consider
transistors and double diodes.

samdolt, 2015-01-12:
I think 200 mils for resistances, capacitors and diode would be nice.

@nickoe and @reportingsjr  : What size do you use in your library?

@CarlPoirier : What the rules to ensure compatibility with old design? Can we just resize and commit the symbol with the same name?

CarlPoirier, 2015-01-12:
There is a cache created for the project, containing the parts used. There
is no obligation about resizing.

I like the size of csmall. I'll think about some rule formulation for it.

CarlPoirier, 2015-01-13:
Here is my suggestion. (Taking care of #101 as well)

3.2. For black-box symbols, pins have a length of 100mil. Large pin numbers can be accomodated by incrementing the width in steps of 50mil.

This way, components with a distinctive shape are not included and might have various pin lengths, but they still have to follow rule 3.1. Resistors and capacitors can thus be 200mil long, such as "csmall".

samdolt, 2015-01-13:
I have made a pull request with some small symbole: https://github.com/KiCad/kicad-library/pull/104

inkblotter, 2015-01-16:
In the real world of kicad schematics parts are whatever size they need to be within that schematic, which may include schematics that have dozens of pages.   Depending on how ones setting for the schematic are, I have used capacitors as small as 40 mils.  Therefore it is silly to create libraries for any of these two pin devices.   A parts wizard should meadiate between those parts that can easily be generated and those fixed size parts that get pulled from libraries, standard or custom.   Ideally, kicad schematics and parts would be in svg and with adaptive pin pitches.  As for the immeadiate issue,  I would never use CP, as it looks like a charge pump to me.  Like others, I have seen CP1 and CP2  used frequently. I would depricate CP and remove it by and by.  I have a slight preference for CP1, being in the US. I associate CP2 with European usage. But either form should be in the 'standard' library, not withstanding my previous comments.

shaneburrell, 2015-03-05:
CP1 is likely the most used, at least in the US. I have only ever seen CP in KICAD and its been a long term annoyance.  CP2 is understandable as well but not what I've seen typically.

CarlPoirier, 2015-03-07:
CP has been eradicated now. Pick whichever of the two symbols left you prefer for your schematics!

closed 2015-03-07 by CarlPoirier