KiCad/kicad-library #101: Note more strictly that prefered length of pins is 100mil

Fat-Zer, 2015-01-11:
KiCad_Library_Convention
> 3.2. Pin has a length of 100mil or more in increments of 50mil if number needs more space.

IMO It would be nice to mark it down more strictly that the pins should be 100mil and shouldn't be increased unless it's really required. The current text may be confused with «100mil or any higher value with 50 mil increment on your taste».

CarlPoirier, 2015-01-11:
Hi,

I'm taking note. I will write a new revision soon and will include your suggestion.

nickoe, 2015-01-11:
Also one should conisder the pin length might not nessesarely align with 50 mil, if the other end ( the one without the circle) should align to some "diagonal" graphic line as in the case of the triangle on the op amps. So the requirement basically is the the wire point should align with the grid.

CarlPoirier, 2015-03-02:
Fixed in the latest KLC revision.

closed 2015-03-02 by CarlPoirier