KiCad/kicad-library #109: move 3d model from unused to relay through hole

andres53016, 2015-01-24:
This 3d models are in the folder unused_3d:
./Relay_ThroughHole_Packages.relay_Finder_34Point51Point7.wings
./Relay_ThroughHole_Packages.Relay_SPDS_OMRON-G6E.wrl
./Relay_ThroughHole_Packages.RELAY_SANYOU_SRD_Series_Form_C.wings
./Relay_ThroughHole_Packages.relay_Finder_34Point51.wrl
./Relay_ThroughHole_Packages.RELAY_SANYOU_SRD_Series_Form_C.wrl
./Relay_ThroughHole_Packages.Relay_SPDS_OMRON-G6E.wings
./Relay_ThroughHole_Packages.relay_Finder_34Point51.wings
./Relay_ThroughHole_Packages.relay_Finder_34Point51Point7.wrl
and the respective footprint are in https://github.com/KiCad/Relays_ThroughHole.pretty, how I can fix this.

michal777, 2015-01-24:
There was some broken links, eg. Relay_SANYOU_SRD_Series_Form_C.kicad_mod lost its 3d model with commit: https://github.com/KiCad/Relays_ThroughHole.pretty/commit/25d6fb88be865b4e6ba77ea21eaaa3cbeeb7eb7f

If you want to fix it, please move the 3d models to kicad-library/modules/packages3d/Relays_ThroughHole/ then rename these .wings/.wrl files to match names of footprints and add relative links to footprints in footprint editor (set proper scale, offset, rotation if necessary).

Before renaming 3d models it would be good to rename footprints (name and reference):
Relays->Relay; Relais->Relay; bistabil->bistable

andres53016, 2015-01-28:
thanks, I create the pull requests

CarlPoirier, 2015-02-15:
Corresponding pull request is #110

CarlPoirier, 2015-02-23:
This got fixed.

closed 2015-02-23 by CarlPoirier