KiCad/kicad-library #115: 3D Module Scaling Errors

stambaughw, 2015-02-03:
The following 3D models require scaling to fit properly on the board:

SMD_Packages/SOT-23-6.wrl
SMD_Packages/SOT-23-5.wrl
SMD_Packages/SOIC-14_N.wrl
Diodes_SMD/Diode-SMA_Standard.wrl

I'm sure there are others as well.  I think it's important for our user's that they not have to rescale the 3D models to fit their boards.

Thanks,

Wayne

CarlPoirier, 2015-02-03:
There is a similar issue on Github. I flagge these as well. No one had
gotten to it, but I'll try soon.

stambaughw, 2015-02-03:
No hurry.  I just would like to have this cleaned up by the next stable
release which hopefully will be around the June/July time frame.

CarlPoirier, 2015-02-23:
Can someone double check SMD_Packages/SOIC-14_N and Diodes_SMD/Diode-SMA_Standard? Both seem fine on my end. I did fix the SOT-23*.

stambaughw, 2015-02-24:
Attached are screen shots of the attached 3d model.  One is scaled to
1/2.54cm and on is unscaled (1).  Both are incorrect AFAICT.  I just did
a git pull and reinstall yesterday unless I did something wrong.

keruseykaryu, 2015-02-24:
I didn't check that recently, but does this issue is still up to date for SOIC-14_N?
https://github.com/KiCad/kicad-library/issues/49

CarlPoirier, 2015-02-25:
I'm sorry Wayne but everything seems fine here. Github plugin activated, kicad-library on master. I don't see your screenshots, so let me upload one.

[image]

SOIC-14_N is alright as well.

stambaughw, 2015-02-25:
> image

I just did a fresh clone using:

git clone https://github.com/KiCad/kicad-library.git

Then I changed the 3D package absolute path to point directly at
SMD_Packages/SOIC-14_N.wrl rather than using CMake to install and I
still get the same thing.  You haven't done a local checkout on your
branch and forgot to merge the changes into the master branch have you?
 Please send me a copy of your SOIC-14_N.wrl file so I can do a diff.

CarlPoirier, 2015-02-26:
No, I did a fresh clone as well! I don't understand either what's
happening. I'll send you the ,wrl and .kicad_mod files for the SOIC-14_N on
your personal email.

CarlPoirier, 2015-03-02:
The issue has been confirmed to not exist anymore with the most recent revision of kicad-library and the latest footprints.

closed 2015-03-02 by CarlPoirier