KiCad/kicad-library #132: Modules cmake

hosier, 2015-02-27:
CMake Error at CMakeLists.txt:5 (install):
  install FILES given no DESTINATION!

CMake Error at CMakeLists.txt:9 (install):
  install FILES given no DESTINATION!

CMake Error at CMakeLists.txt:17 (install):
  install DIRECTORY given no DESTINATION!

CMake Error at CMakeLists.txt:22 (install):
  install DIRECTORY given no DESTINATION!

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.1)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!

hosier, 2015-02-27:
Output errors from previous message are unrelated. I'm trying to add the modules and thought that I needed to run cmake somewhere.

keruseykaryu, 2015-02-28:
Is it means that your issue report is invalid for kicad-library? If yes, please close it ASAP.

closed 2015-03-03 by hosier