KiCad/kicad-library #135: Author agrees to use his work

timofonic, 2015-03-05:
Hello.

I'm not sure if this is the best way to point to this, probably not.

The author of these libraries agreed to be merged to the official ones if there's license compatibility.

http://smisioto.no-ip.org/elettronica/kicad/kicad-en.htm

> Do you plan to merge your efforts with the official library available on
GitHub? It would be great.
> No, I really don't plan to. My libraries are released under dual
licensing, GPL and CC-BY-SA, so if the developers wish it, they can just
merge them into their repos.

There's lots of KiCad libraries, but they are on different firms. I would love that KiCad can have a very complete library, but I know managing a big library and make people to merge can be a very difficult task.

Regards.

shaneburrell, 2015-03-05:
Is this going to happen? Anyone working on it?

michal777, 2015-03-05:
Some of the libraries are in official KiCad libraries (but 3d models could be upgraded), some of them can be added and others may conflict with our libraries (we don't want duplicates). It's a lot of work to add this content without creating much bigger mess than we have. If somebody will work on it probably the libraries or their parts will be added one by one, not all. Then they will require some corrections according to kicad library convention and to link 3d models properly.
I don't know if there is somebody working on it, probably not.

SchrodingersGat, 2015-11-09:
I have adapted a few of the footprints and models from his website to match KLC and such.

Terminal Blocks
A number of the terminal blocks have been added here:
https://github.com/KiCad/Terminal_Blocks.pretty
Pheonix PT
Pheonix MPT
Pheonix MKDS

Socket strips:
The 2.00mm pitch socket strips have been added to the Socket_Strips.pretty repo (pull request currently open)
https://github.com/KiCad/Socket_Strips.pretty/pull/4

Pin Headers:
Same as above:
https://github.com/KiCad/Pin_Headers.pretty/pull/10

Hirose DF13 connectors
https://github.com/SchrodingersGat/Connectors_Hirose.pretty
Note: This is at the moment just in my personal repo. I am happy for this to be used by github.com/KiCad/ freely

SchrodingersGat, 2015-11-09:
Oh and also, my JST library

https://github.com/SchrodingersGat/Connectors_JST.pretty

Same as above. Very happy for this to be integrated into master repo. Also happy to assist with KLC compliance as required.

pointhi, 2015-11-24:
For JST I want to say there is already a pull request for all JST PH and JST SH connectors, made by me: #296

SchrodingersGat, 2016-12-06:
Closing due to inactivity

closed 2016-12-06 by SchrodingersGat