KiCad/kicad-library #153: How to resolve mess and hotchpotch in `special.lib`

keruseykaryu, 2015-03-17:
I think `special.lib` is a type of bag in which very strange things were thrown. 
Here are my proposals how to clean up this mess and finally remove this problematic bagagge. 
However, I still have a lot of doubt where to put some devices indicated by '???'.

| Device | Move to | Info
|---|---|---
| 20RS10 | lattice.lib | New library
| 74LS629 | 74xx.lib
| 754C923 | ??? | 
| 75LBC176 | interface.lib
| 9346 | memory.lib
| AD725 | video.lib
| ADUC816 | microcontrollers.lib
| AV9173 | video.lib
| CX7930 | ??? | 
| DG884 | ??? | 
| DP83848I | interface.lib
| DS1267 | maxim.lib | New library
| DS1302 | maxim.lib | New library
| DS1307+ | maxim.lib | New library
| DS1307N+  | maxim.lib | New library
| DS1307Z+ | maxim.lib | New library
| DS1602 | maxim.lib | New library
| DS1621 | maxim.lib | New library
| DS1804 | maxim.lib | New library
| DS7820 | interface.lib
| DS7830 | interface.lib
| DS8830 | interface.lib
| DS90C124 | interface.lib
| DS90C241 | interface.lib
| EL2001 | interface.lib
| EL7242C | interface.lib
| EP1210 | altera.lib | New library
| EP1810 | altera.lib | New library
| EP300 | altera.lib | New library
| EP310 | altera.lib | New library
| EP320 | altera.lib | New library
| EP600 | altera.lib | New library
| EP910 | altera.lib | New library
| FD1771 | interface.lib
| FD2791 | interface.lib
| FD2795 | interface.lib
| HA2540 | linear.lib
| HA5033 | linear.lib
| HCLD0438 | display.lib
| HCPL2731 | opto.lib
| HD63484 | video.lib
| HD63484_PLCC | video.lib
| HI524 | ??? | 
| HM53461 | memory.lib
| HM53462 | memory.lib
| ICL7660 | dc-dc.lib
| ICM7209 | ??? | 
| IDT7132 | ??? | 
| IDT7201 | ??? | 
| IDT7202 | ??? | 
| IQXO/7X | ??? | 
| IRF7404 | transistors.lib
| LCD4 | display.lib
| LIGNE_A_RETARD | device.lib
| LM16255K | display.lib
| LM1881 | video.lib
| LM35 | sensors.lib
| LM5051 | power-management.lib | New library
| LT1080 | interface.lib
| LTC1257 | ??? | 
| LTC4007 | ??? | 
| M3003 | ??? | 
| MAX232 | interface.lib
| MAX202 | interface.lib
| MAX2606 | ??? | 
| MAX2505  | ??? | 
| MAX2507 | ??? | 
| MAX2508 | ??? | 
| MAX2509 | ??? | 
| MAX310 | ??? | 
| MAX311 | ??? | 
| MAX3226 | interface.lib | 
| MAX453 | ??? | 
| MAX5436 | maxim.lib | New library
| MAX6355 | ??? | 
| MAX666 | ??? | 
| MAX777 | ??? | 
| MB81461ZIP | memory.lib
| MB88303P | ??? | 
| MC1413 | interface.lib | 
| ULN2003  | interface.lib | 
| ULN2003A | interface.lib | 
| MC1413P | interface.lib | 
| MC1417 | interface.lib | 
| MC33064D | ??? | 
| MC34064D | ??? | 
| MC34064P | ??? | 
| MC33064P | ??? | 
| MC3486 | interface.lib
| MC3487 | interface.lib
| MK4501 | ??? | 
| MK4511 | ??? | 
| MT43C4257 | interface.lib
| MT43C4258 | interface.lib
| Max691 | ??? | 
| PAL16L8 | lattice.lib | New library
| PAL20L8 | lattice.lib | New library
| PCI9030-PQFP176 | conn.lib | 
| S5933_PQ160 | interface.lib
| SA605D | ??? | 
| SI582 | ??? | 
| SI9986 | interface.lib
| SRV05 | ??? | 
| S_178 | ??? | 
| Si570 | ??? | 
| Si571 | ??? | 
| TC524256Z | memory.lib
| TCXO3 | ??? | 
| TCXO_DFAS11 | ??? | 
| TDA1950 | video.lib
| TDA1950F | video.lib
| TDA7260 | video.lib
| TDA_2593 | video.lib
| TDA_9500 | video.lib
| TDA_9503 | video.lib
| TDA_9513 | video.lib
| TDC1408 | ??? | 
| TEA2014 | interface.lib
| TEA5115 | video.lib
| TIL111 | opto.lib
| TL431_SOT-23-3| REMOVE | Already exist
| TL431_TO92 | REMOVE | Already exist
| TL7702 | ??? | 
| TL7705 | ??? | 
| TMS320LF2406 | texas.lib
| TMS4C1050N | ??? | 
| TPD2S017 | ??? | 
| TPD4S1394 | ??? | 
| TQSMT-35 | ??? | 
| U2270B | interface.lib
| UA9637 | interface.lib
| UA9638 | interface.lib
| UVC3100 | ??? | 
| UVC3101 | ??? | 
| X9118 | xicor.lib | New library
| X9250 | xicor.lib | New library
| X9258 | xicor.lib | New library
| X9258-PART | xicor.lib | New library
| Z8530 | zilog.lib | New library

Do you have any other proposals or have different solutions than mine?

ricardocrudo, 2015-03-18:
What about PMIC (power management integrated circuit) for: MC33064D,MC34064D,MC34064P,MC33064P,TL7702,TL7705

keruseykaryu, 2015-03-18:
> What about create pmic.lib (power management integrated circuit) for: MC33064D,MC34064D,MC34064P,MC33064P,TL7702,TL7705

I have suggested new `power_management.lib` library to which these parts can be moved to. IMO, we have to avoid shortcuts - as possible - in library names.

ricardocrudo, 2015-03-19:
more suggestions:
SA605D: rfcom
SRV05, TPD2S017, TPD4S1394: esd_protection.lib or transient_voltage_suppression.lib (TVS)
Si570, Si571: oscillators.lib or programmable_oscillators.lib to be more specific.
Actually, maybe, clock_generators.lib could be oscillators.lib.

keruseykaryu, 2015-03-19:
`esd_protection.lib` sounds reasonable. `oscillators.lib' - Short and pithy.

keruseykaryu, 2015-03-21:
> more suggestions:
> SA605D: rfcom

This is an FM radio component. IMO this is not a strict RF data communication device.

nickoe, 2015-03-22:
So maybe rf_mixers for the SA605D?

keruseykaryu, 2015-03-22:
I'm looking for some more universal name for radio devices, also for the future symbols that might appear.

nickoe, 2015-03-22:
So what about radio_devices or rf_devices?

CarlPoirier, 2015-03-22:
Hi Kerusey,

Thanks for taking care of this.

Here is some more information on unknown parts:

S178 is a video pulse generator http://pdf.datasheetarchive.com/indexerfiles/Scans-052/DSAIH00054578.pdf
SI582 might be a video amplifier but I'm not sure at all.
M3003 is actually M3002 http://www.mixinc.net/atari/datasheet/M3002.pdf

ChlorideCull, 2015-03-31:
I find some battery chargers in `regul.lib`, which isn't really the place for them (they don't strictly do only regulating). Splitting them out to a `power_management.lib` would be a really nice solution.

keruseykaryu, 2015-03-31:
> I find some battery chargers in regul.lib, which isn't really the place for them

Would you tell us which they are?
Your #168 also is a candidate to put it in the different library.

keruseykaryu, 2015-03-31:
> @CarlPoirier: SI582 might be a video amplifier but I'm not sure at all.

I have found one DS (http://www.farnell.com/datasheets/1748226.pdf) where Si582 is referenced as video amplifier.

The destination list is now almost complete. Please review once more.

ChlorideCull, 2015-04-01:
>Your #168 also is a candidate to put it in the different library.

Yeah, I placed it for there since the others are there, and as of right now there isn't a better place to put them. They can be moved when there is. The parts in `regul` as of right now are:

* MAX1647 (Battery Charger)
* MAX1648 (Battery Charger)
* BQ2003 (Controleur de chargeur de batteries)

CarlPoirier, 2015-04-02:
@keruseykaryu It looks great on my end. Can we get this merged before we proceed to rename the libraries with capital letters? I talked about April 8th in the other discussion #161 , but I think we should prioritize this. Are there any merges that must go through before this one?

ChlorideCull, 2015-04-02:
I've taken the liberty to move over the battery charger and #168 into `Power_Management.lib`. Pull request #171.

keruseykaryu, 2015-04-02:
> Can we get this merged before we proceed to rename the libraries with capital letters?

Sorry Carl, but I can not find some spare time to split this library.

> Are there any merges that must go through before this one?

#152 #145

CarlPoirier, 2015-04-06:
I myself don't have much time either. I suggest we undertake this bits by bits and write in the Info column whenever some parts are done.

ricardocrudo, 2015-04-14:
I did a script that can be helpful in this work of moving parts. It's in my repo https://github.com/ricardocrudo/kicad-utils and is called move_part.py, inside of schlib directory. The output library file will probably have other modified parts, besides of the "move" content. Because the script uses other algorithm to generate library files.

CarlPoirier, 2015-04-25:
This should be it in #181. Can anyone go over the changes?

ricardocrudo, 2015-04-29:
Hmm, ok. I'll check this flag. Thanks by report.

I've checked Carl's request, each part. Everything is OK, ready to merge.

CarlPoirier, 2015-04-30:
Thanks Ricardo.

closed 2015-04-30 by CarlPoirier