KiCad/kicad-library #155: Diodes symbols

ricardocrudo, 2015-03-18:
I started to working in changing the diode symbols, as it was said here: https://github.com/KiCad/Diodes_SMD.pretty/pull/6
Now I have a doubt about the D_Schottky_x2_KCom and D_Schottky_x2_Serial diodes.
The current pinout for those symbols are, respectively:
A1,K,A2 <-> 1,2,3
A,AK,K <-> 1,2,3
But, I've used parts which have the following pinout: 
A1,K,A2 <-> 1,3,2
A,AK,K <-> 1,3,2
My question is how to manage this, create new symbols with this alternative pinout? If so, how to name them?

Now talking about all diodes, i.e. regular diode, schottky, zener..., would be better use letters in the pin number field instead of numbers? In this way I think is possible to solve the pinout variations and prevent possible mistakes.

michal777, 2015-03-18:
Letters in pin numbers would break rule 6.8. AFAIK for general symbols there should be more symbols with different pin numbering. Look at transistors in device.lib ("Q_..."). 3-pin packages have their standard pin numbering so don't care about schematic symbols for them. 2-pin diodes should have pin 1 assigned to cathode. Then they will match standard numbering of DO-... packages and two pin TO-... packages (it seems that most of / all manufacturers use pin 1 of these packages as cathode).

CarlPoirier, 2015-03-22:
Exactly.

CarlPoirier, 2015-04-24:
Everything has been merged. @keruseykaryu, can you announce the change on the users group?

keruseykaryu, 2015-04-24:
Done.
https://groups.yahoo.com/neo/groups/kicad-users/conversations/messages/19990

CarlPoirier, 2015-04-24:
Thanks! I'm closing this issue now.

closed 2015-04-24 by CarlPoirier