KiCad/kicad-library #162: Move 3D packages into .pretty repos

alexforencich, 2015-03-25:
Having the 3D packages and footprints in separate repos is a recipe for Bad Things.  It would be much easier to manage footprints and 3D packages if the footprints and corresponding packages are managed in the same repo.  Note that this will require modifications to how Kicad locates 3D packages - it must me modified to also search for the 3D package relative to the footprint file first.  This will also enable 'self-contained' footprint libraries with 3d packages to be used without being forced to use absolute paths or install them with the system kicad library.

CarlPoirier, 2015-03-26:
Hi,

There was a conversation about this subject a while ago. Have a read here:

https://lists.launchpad.net/kicad-developers/msg14793.html

Fat-Zer, 2015-05-17:
What about moving it out from the kicad-library repo? 
IMO It doesn't make a lot of sense to store them nearby the schematic...

Bitbored, 2016-01-05:
I agree with this. I think it makes some sense to keep footprints and 3d packages together, since they are related most of the time (apart from some strange parts with a common footprint but a weird 3D shape).

I understand there might be good reasons to keep the 3D files and foortprints separate, but having the 3D files and symbols together in the same repo doesn't make a lot of sense.

Can we continue this discussion here, or is the mailing list or launchpad the preferred channel?

alexforencich, 2016-01-05:
There are a whole bunch of architectural issues in kicad that need to be fixed, and this is one of them.  Or at least related to it.  There is no search path for 3D packages right now, or at least the last time I checked.  Kicad only looks in one spot for the packages, this needs to be extended so that several locations are tried (e.g. same folder as the footprint in question).  Until this is fixed, the 3D packages cannot be moved.

reportingsjr, 2016-01-05:
I believe Cirilo Bernardo's work (
https://code.launchpad.net/~cirilo-bernardo/kicad/3d_initial_merge ) should
solve that issue Alex. Hopefully it will be merged in the next month or two.

alexforencich, 2016-01-05:
Looking forward to it.  Kicad has a lot of promise, but also a lot of issues.

Shackmeister, 2017-06-13:
@alexforencich Hi Alex, a seperate packages3D repo has now been made and is on it's way to be supported in kicad directly. According to KLC $kisysmod3D is now being added to all footprints. 
Can you please close this issue if its not relevant any more?

Shackmeister, 2017-07-01:
@alexforencich ill close this now, please let me know if you think it shouldn't be

seppestas, 2017-07-04:
Mmh, this issue states "move 3D packages into .pretty repos", not "move 3D packages in separate repo", so IMO the discussion is still open...

I understand moving the 3D packages in the .pretty repos requires a big architectural overhaul of KiCad's 3D model system and probably of the footprint system, but I think it still makes sense since the footprints reference the 3D models and typically footprints and 3D models are pretty related.

There are a lot of benefits in having the 3D models in a separate repo though. This could make it easier to override the 3D model for a footprint in different projects.

An example case: I have a design that uses [Neutrik NMJ4HCD2](http://www.neutrik.com/en/audio/plugs-and-jacks/m-series/nmj4hcd2) connectors. This connectors has the same footprint as the [Neutrik NMJ4HFD2](http://www.neutrik.com/en/audio/plugs-and-jacks/m-series/nmj4hfd2) connector but a different 3D model. It would be nice if KiCad would easily allow changing the 3D model without changing the 3D model so I can easily switch to using the NMJ4HFD2.

I have a few ideas on how this use-case could be supported:

- Enumerate compatible 3D files in the footprint and make the project/layout file store wich one should be used. Currently it is possible to list multiple 3D files per footprint, but KiCad combines them instead of giving the option to select one.
- Use a more generic name for the 3D file and change the path(s) to the location of the 3D files on a per-project basis (e.g by using an environmental variable in the 3D file path that can be set in the project). This is where having the 3D models in a separate repo makes sense.

poeschlr, 2017-07-04:
As far as i understand it, in future the number of repos will be reduced drastically. We will (hopefully) move away from the one repo per lib nonesense.
In such a case including the 3d files in the same repo as the footprints only makes sense if we also include the symbols and basically everything else into the same repo. (everything else would confuse users because they do not know how to find stuff.)
My vote would be on: One repo for symbols, one for footprints, one for 3d models, one for templates, ...

Issue https://github.com/KiCad/kicad-library/issues/1360 evolved into discussing exactly that.

seppestas, 2017-07-04:
Personally I prefer the multi-repo approach, I just think it misses some kind of automatic way to download / version pin the repositories. The current implementation of the Github downloader is fairly inconsistent and doesn't really work when using anything but the official KiCad libraries.

Having multiple smaller repo's allows to more easily fork the repo's you want to add custom stuff to while allowing you to keep using the original repos for other footprints.

I don't see why having the symbols in the same repo is a requirement for having the 3D files in the footprint repos. For my personal projects I typically don't even use the symbols anymore thanks to [SKiDL](https://github.com/xesscorp/skidl).

alexforencich, 2017-07-04:
Honestly, I don't really care where the 3D packages go.  It just makes no sense to keep them in the same repo as the symbols when the footprints are in their own repo, considering the 3D packages are referenced by the footprints and not the symbols.  

And in terms of version pinning, that should be avoided entirely by switching to an Eagle like library format where the symbols and footprints are physically stored in the same file, eliminating the potential for any discrepancies between the footprints and symbols.

SchrodingersGat, 2017-07-04:
> And in terms of version pinning, that should be avoided entirely by switching to an Eagle like library format where the symbols and footprints are physically stored in the same file, eliminating the potential for any discrepancies between the footprints and symbols.

Whether or not this would be a good idea, this is not the way that the KiCad libraries work now or will work in the future. I don't think there's much sense trying to debate this in a post here. 

> Mmh, this issue states "move 3D packages into .pretty repos", not "move 3D packages in separate repo", so IMO the discussion is still open...

The 3D models will only (in future) be located in the `packages3d` repository. This issue is definitely closed :)

closed 2017-07-04 by SchrodingersGat

poeschlr, 2017-07-05:
@seppestas wrote:
> Personally I prefer the multi-repo approach, I just think it misses some kind of automatic way to download / version pin the repositories.

Be a librarian for a week and tell us if you still like it ;)

shaise, 2017-07-29:
I really think seppestas request is extremely useful and reasonable. As others, I maintain a library of my own with footprints not available in the default library. Having the 3D models inside the pretty folder, makes it much easier when I want to share my library with others, or when I edit my project on a different PC.