KiCad/kicad-library #163: Library arduino

pouic13, 2015-03-25:
Hello evryone,<br> I made a library who contain most popular arduino board. It's contain .mod, .pretty, 3D views and shematics. It must be corrected but it exit.<br>You can download the full library on kicad forum URL: https://forum.kicad.info/t/library-for-shield-arduino/482 <br> Or on my blog URL: http://meta-blog.eklablog.com/kicad-librairie-arduino-pretty-p930786 <br> On my blog you have a prewiew that contain the library.<br>This is the first time i use github and i don't understan how it workd and to cap it all i'm french. So be indulgent :)

michal777, 2015-03-25:
In this thread:
https://forum.kicad.info/t/library-for-shield-arduino/482/8
I suggested creating issue because I don't know much about arduino and templates in kicad but I guess that it would be better to make templates based on this library and it should be discussed here.

keruseykaryu, 2015-03-26:
> (...) it would be better to make templates based on this library (...)
Exactly. Templates were designed just for that.
KiCad Manager manual gives the guidelines how to prepare complete template.

pouic13, 2015-03-26:
ok i'will watch how to creat a template.

keruseykaryu, 2015-03-26:
> But where i found " KiCad Manager manual " ?

Open KiCad and call Help -> Contents. Page 6.

pouic13, 2015-03-26:
I don't have file in Help. I use arch linux and i install paquet kicad-bzr and kicad-library-bzr. But in /usr/share/kicad/ i have only /demos, /library, /modules, /template and not /help or looks like help directory. Can I find it at an another place?

nickoe, 2015-03-26:
http://www.kicad-pcb.org/display/KICAD/KiCad+Documentation

pouic13, 2015-03-26:
Thank nickoe. <br>But just to be sure, i do made one template per one shield ?<br> so for my library: 7 for shield with no contour + 7 for shield with contour ?

michal777, 2015-03-26:
I'd suggest to create 7 templates, one for each shield. Contour should be drawn on Edge.Cuts layer as it is in template of shield for raspberry.

pouic13, 2015-03-26:
ok rooo f**k (sorry)

pouic13, 2015-03-30:
Hello, i start to redraw my library and i want to know, how i do orginise my project for summit a pullrequest? My lybrairie Template have got this structure: in file Template you have 7 directorys who each contain 1 complete board. i create a directory for modules who i use in template and i have an other directory for 3D view in relation whith modules. I keep this structure? or do I place directory "module" and "3D" in directory of "Template of each board" ?  An other question what size "edge.cuts" an "Dwgs.user" do have?

michal777, 2015-03-30:
> An other question what size "edge.cuts" an "Dwgs.user" do have? 
There are no rules but I think 0.15 mm will be ok (the same as silkscreen).

You don't need to attach *.pretty footprints because they can be extracted from *.kicad_pcb (board).
You don't need to attach schematic library but there should be correct -cache.lib in project directory.
It's possible to use ${KIPRJMOD} to use 3d models existing in project directory. Example of path in *.kicad_pcb file:
```
    (model ${KIPRJMOD}/Socket_Arduino.3dshapes/Socket_header_Arduino_1x08.wrl

pouic13, 2015-03-30:
Thank micha777 it's work great for 3D views.
I place anyway *.pretty. I think it's good to keep a directory in template where you can change/place other footprint if they are propose do not correspond. 
Ok know my first board is ready! i have correct edge cuts and meta are ok i'will try to summit a pull request...

CarlPoirier, 2015-04-24:
The templates for the Arduino boards made by @pouic13 have been merged.

closed 2015-04-24 by CarlPoirier

giogziro95, 2017-05-25:
Hi, everyone.

Which library has this been added to? I can't find one. It's not available in my KiCad install either.

@jonathan-iap Thanks for your awesome work.

giogziro95, 2017-05-25:
Oh, [here they are](https://github.com/KiCad/kicad-library/tree/master/template).