KiCad/kicad-library #164: DIN 5-pin connector with non standard numeration

ricardocrudo, 2015-03-26:
Looking at the DIN_5 symbol from conn.lib I've checked that the pins numbers are numerated sequentially but the standard DIN 5-pins connectors used to have another numeration way, like the following references:
http://www.mouser.com/ds/2/670/sds-xxj-535158.pdf
http://www.mouser.com/ds/2/209/KC-301191-193300.pdf

Is this a bug or desired feature? I'm needing a standard 5-pin din connector, thus I'd like to know if I fix this one or keep it and create another one.

samdolt, 2015-03-30:
I didn't found any corresponding footprint in the library. Did you find one?

keruseykaryu, 2015-03-30:
[image]

keruseykaryu, 2015-03-30:
This symbol doesn't have any corresponding footprint. To be in accordance with the standard pinning, we need to improve that symbol. Even if this connector type is a bit obsolete.

CarlPoirier, 2015-04-01:
Can you find a DIN_5 with that pinout? If not, then you can fix this one.

keruseykaryu, 2015-04-03:
I'm pretty sure we do not find this pin arrangement. 
IMO this is one of the results of joyful productivity from some user who never saw German DIN 41524.

CarlPoirier, 2015-04-06:
Fixed in latest pull request from Ricardo. Thanks!

closed 2015-04-06 by CarlPoirier