KiCad/kicad-library #169: Missing SMD diode footprints

cb88, 2015-04-01:
Currently 1206, 1210, 805, 603, parts etc... are missing. Currently on my boards I just use a resistor footprint however this lacks polarity indication for manufacturing.

samdolt, 2015-04-01:
In SMD_Packages.pretty, you can find
* SMD-1206_Pol.kicad_mod
* SMD-1210_Pol.kicad_mod
* SMD-2010_Pol.kicad_mod
* SMD-2112_Pol.kicad_mod
* SMD-2512_Pol.kicad_mod
* SMD-4032.kicad_mod

0403, 0805 and 0603 seems to be missing.

samdolt, 2015-04-01:
@keruseykaryu I propose to delete these file and to put a copy in to Diodes_SMD and Capacitors_SMD with CP_* and D_* name. I think footprint filter in device.lib should be already correct, but it's better to verify.

I can handle that later this week if you're ok.

keruseykaryu, 2015-04-03:
> I propose to delete these file and to put a copy in to Diodes_SMD and Capacitors_SMD with CP_* and D_* name.

I think you should ask Carl or Michal. Footprint libraries isn't my major interest here.

samdolt, 2015-04-03:
@CarlPoirier What do you think?

michal777, 2015-04-03:
The SMD-****_Pol.kicad_mod can be moved if these polarised diodes and capacitors really exist.
Polarised tantalum capacitors are in separate library and they have different naming standard.
Diodes in this kind of packages are not popular (I found in some shop: 0603, 0805 and 1206, bigger diodes are usually SOD..., DO..., SM* etc).
What's more, they should be improved to meet kicad library convention and IPC-7351C.

cb88, 2015-04-03:
Diodes like this are useful for small boards and are always polar... Think Arduíno nano etc.....

Also on my boards I just standardized on 805 and 603 partes where i could as these are about as small of a discret component as I care to solder.

All my caps this size are ceramic so wouldn't matter.

michal777, 2015-04-03:
Are you sure the diodes aren't minimelf, micromelf or SOD...?

cb88, 2015-04-03:
Sure .... But that doesn't mean you can't get them in standard sizes. Plenty non melf or sod parts on digikey or Newark etc...

CarlPoirier, 2015-04-08:
A quick look on Digikey reveals zener diodes in 402, 503, 603, 805, 1005, 1206. 

The packages can be moved and eliminated from SMD_Packages.pretty. One step closer to remove this to vague library!

Shackmeister, 2017-06-12:
@cb88 I believe this has been solved since, could you close this issue if you agree?

cb88, 2017-06-12:
Yes it appears that it has been resolved thought I am currently using a resistor + silkscreen polarity indication manually (tisk tisk) . As the standard sized SMD components now exist in Diodes_SMD.

closed 2017-06-12 by cb88