KiCad/kicad-library #172: Part 74LS51 in 74xx.lib has wrong symbol and wrong pin assignment

onitake, 2015-04-09:
The circuit library 74xx.lib shipped with KiCAD contains an entry for the 74LS51 logic gate part. According to the respective data sheet, this part is wrong. Neither the gate symbol nor the pin assignment reflect the actual part.

74xx.lib shows the part as a dual 2x2 input AND/OR combined gate, while the 74LS51 contains one 3x2 and one 2x2 gate. Only the 7451 and 74S51 parts do not have the additional inputs. Also, the output pin of part B is number 8, not 9 as in the library.

Strictly speaking, part A and part B also swapped, but this has no functional relevance.

Texas Instruments offers a data sheet for this part here: http://www.ti.com/lit/ds/symlink/sn74ls51.pdf

onitake, 2015-04-09:
EESchema-LIBRARY Version 2.3  Date: Thu 09 Apr 2015 02:01:50 CEST
#encoding utf-8
#
# 74LS51
#
DEF 74LS51 U 0 10 Y Y 2 F N
F0 "U" 200 250 60 H V C CNN
F1 "74LS51" 300 -250 60 H V C CNN
F2 "~" 0 0 60 H V C CNN
F3 "~" 0 0 60 H V C CNN
$FPLIST
 DIP-14*
$ENDFPLIST
DRAW
X GND 7 -50 -275 0 U 60 60 0 0 W N
X VCC 14 -50 275 0 D 60 60 0 0 W N
A -50 -150 125 -899 899 0 1 0 N -50 -275 -50 -25
A -50 150 125 -899 899 0 1 0 N -50 25 -50 275
A 75 0 146 -590 590 0 1 0 N 150 -125 150 125
A 170 -148 274 941 327 0 1 0 N 150 125 400 0
A 170 148 274 -941 -327 0 1 0 N 150 -125 400 0
P 4 0 1 0  -50 -25  -150 -25  -150 -275  -50 -275 N
P 4 0 1 0  -50 25  -150 25  -150 275  -50 275 N
P 4 0 1 0  75 -150  125 -150  125 -75  200 -75 N
P 4 0 1 0  75 150  125 150  125 75  200 75 N
X ~ 1 -450 250 300 R 50 50 1 1 I
X ~ 8 700 0 300 L 60 60 1 1 O I
X ~ 9 -450 -50 300 R 60 60 1 1 I
X ~ 10 -450 -150 300 R 60 60 1 1 I
X ~ 11 -450 -250 300 R 60 60 1 1 I
X ~ 12 -450 150 300 R 60 60 1 1 I
X ~ 13 -450 50 300 R 60 60 1 1 I
X ~ 2 -450 250 300 R 50 50 2 1 I
X ~ 3 -450 50 300 R 60 60 2 1 I
X ~ 4 -450 -50 300 R 60 60 2 1 I
X ~ 5 -450 -250 300 R 60 60 2 1 I
X ~ 6 700 0 300 L 60 60 2 1 O I
ENDDRAW
ENDDEF
#
#End Library

ricardocrudo, 2015-04-29:
@onitake I just commit a fix using your symbol plus small adjusts. Thanks for reporting.

The symbol is violating the KLC but I did commit because it's bug.

closed 2015-04-29 by ricardocrudo

onitake, 2015-04-30:
I did not know about the [KLC](https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention).

Thanks for pointing that out, I will try to follow it in the future.

felipesanches, 2016-06-28:
I also noticed this bug when looking for the 74LS50 which is missing in KiCAD. Should I report an issue about it separately?

Also, I'm using the stable debian kicad package in Trisquel GNU/Linux. In order to make my work easily accessible to users, I'm not willing to make my schematics only viewable with a git build of Kicad. To acchieve that, should I add the missing symbol as a custom library in my project's git repo?

felipesanches, 2016-06-28:
oh! I just noticed that the part I need is actually the 7451 instead of the 74LS51. The difference is that the 7451 does not have the third AND-input in the first combination of gates. The 7451 is still lacking in kicad libraries.

felipesanches, 2016-06-28:
My drawing of the 7451 is available at: https://gitlab.com/fsanches/PatinhoFeio_KICAD/tree/master/libs

diggit, 2016-07-06:
>Also, I'm using the stable debian kicad package in Trisquel GNU/Linux. In order to make my work easily accessible to users, I'm not willing to make my schematics only viewable with a git build of Kicad. To acchieve that, should I add the missing symbol as a custom library in my project's git repo?

Hi @felipesanches, once you add symbol to schematic, symbol is saved in cache. When symbol from source library becomes unavailable, KiCad tells you about it and switches to cached par AFAIKt. You can ship source library with project too...