KiCad/kicad-library #173: wrong 3d model

andres53016, 2015-04-09:
The 3d models Diode_LED_3mm_round_vertical and Diode_LED_5mm_round_vertical In kicad-library/modules/packages3d/Diodes_ThroughHole.3dshapes/ are DO-35 and the LED have own directory.
Can I delete this?
And replace the current 3d model for the mentioned in issue #135

CarlPoirier, 2015-04-19:
The 3d model path should be placed in a folder named the same as the pretty library. What do you mean by "the LED have own directory" ?

michal777, 2015-04-19:
@CarlPoirier
The 3d models assigned to the footprints are wrong (DO-35 instead of 5mm led) and the LEDs in Diodes_ThroughHole.pretty are duplicates of the leds that are in LEDs.pretty.
So I think it's good idea to remove the duplicates with wrong models.

CarlPoirier, 2015-04-19:
Aah I wasn't sure what he meant with the directory issue. Indeed, @andres53016, go ahead and we'll approve your pull request.

andres53016, 2015-04-20:
Hi I created #178 https://github.com/KiCad/LEDs.pretty/pull/4 https://github.com/KiCad/Diodes_ThroughHole.pretty/pull/7

CarlPoirier, 2015-05-02:
This issue is fixed. Thanks @andres53016.

closed 2015-05-02 by CarlPoirier