KiCad/kicad-library #176: Discret:TO92DGS has pins ordered SGD

strykeroz, 2015-04-15:
Hi,

Please correct me if I'm wrong, but this appears to have D & S pins swapped.
[image]

Geoff

ricardocrudo, 2015-04-15:
Where does this footprint come from? Which .pretty directory?

strykeroz, 2015-04-15:
According to my library table, ${KIGITHUB}/Discret.pretty should be the source.
This is from a recent launchpad build, 2015-04-09 BZR5589

ricardocrudo, 2015-04-15:
It's named as DGS, since the square pin is the first, it seems right, though, it's violating some KLC rules.

michal777, 2015-04-15:
TO-92 packages that meet KiCad Library Convention rules are in: Housings_TO-92.pretty
This one is going to be removed (when somebody will find some time to sort out Discret.pretty).

strykeroz, 2015-04-15:
The schematic symbol for the BS170 MOSFET in the transistors library has pads D-G-S however all TO-92 packages in Housings_TO-92.pretty currently have pads 1-2-3.  Choosing any of those footprints results in no ratsnest to the BS170 footprint.

michal777, 2015-04-15:
This is because schematic symbol is also bad and should be improved to use standard numbering.

ricardocrudo, 2015-04-15:
The BS170 that I have here, from transistors lib, is numbered from 1 to 3. Did you install your libraries manually or from your distro repo (in the case that you are using linux)?

strykeroz, 2015-04-15:
Interesting.  This was a version made last week using the repo through kicad-winbuilder.  I'll update and report here if it changes.

ricardocrudo, 2015-04-15:
Is that kicad-winbuilder for libraries or the KiCad program?

strykeroz, 2015-04-15:
It downloads everything.  A search for transistors.lib only shows the instances in the source files it has downloaded and the KiCad directory tree it creates.  There are no other copies on this PC.  I will archive the KiCad build I used for the above, and rebuild.

strykeroz, 2015-04-15:
I archived the existing build to a 7z, deleted the directory and rebuilt.  No errors but the version number curiously remains the same.  As does the component in eeschema:
[image]

strykeroz, 2015-04-15:
Further, as expected, when importing .cmp into Pcbnew, the following errors:
*** Error: Component 'Q1' pad 'D' not found in footprint 'Housings_TO-92:TO-92_Molded_Wide_Oval' ***
*** Error: Component 'Q1' pad 'G' not found in footprint 'Housings_TO-92:TO-92_Molded_Wide_Oval' ***
*** Error: Component 'Q1' pad 'S' not found in footprint 'Housings_TO-92:TO-92_Molded_Wide_Oval' ***

ricardocrudo, 2015-04-15:
That's odd. Could you paste your netlist and cmp files at pastebin?

keruseykaryu, 2015-04-16:
@strykeroz: Please remove project library cache, reopen schematic, recreate netlist and pass through CvPcb again.

strykeroz, 2015-04-16:
@keruseykaryu you are correct.  This is now solved as being my project library cache issue.

closed 2015-04-16 by strykeroz