KiCad/kicad-library #177: USB symbols and footprints

ricardocrudo, 2015-04-15:
I'd like to suggest a little change in the USB symbols. Currently there are 3 symbols:
* USB
* USB-MICRO-B
* USB-MINI-B

My suggestion is keep only 2 USB symbols: 
* USB (new symbol)
* USB-OTG

The new USB symbol would be a simple square with 5 pins: VCC, D-, D+, GND and Shield, numbered 1 to 5. It could be used to both, USB Type A or USB Type B.

The new USB-OTG would be a simple square with 6 pins: VCC, D-, D+, ID, GND and Shield, numbered 1 to 6. It could be used to USB mini or micro.

If you agree with this changes I can pull request them including the changes in the footprints.

keruseykaryu, 2015-04-16:
I have no objections.

CarlPoirier, 2015-04-19:
It's a good idea.

CarlPoirier, 2015-05-02:
Thanks Ricardo.

closed 2015-05-02 by CarlPoirier