KiCad/kicad-library #179: SOT-23 footprints in Diodes_SMD appear to have incorrect spacing.

stambaughw, 2015-04-22:
I was reviewing a board layout that I just did and noticed that the spacing between pins 1 and 2 is too narrow for footprints Diode_SOT23 and Diode_SOT23_handsoldering in the Diodes_SMD library.  The current spacing is 1.6mm but every document I could find specifies 1.9mm.  Both footprints should be reviewed for accuracy.

ricardocrudo, 2015-04-22:
Indeed. I can fix this in the pull request: KiCad/Diodes_SMD.pretty#6

ricardocrudo, 2015-04-22:
I just remember that SOT-23 is duplicated in KiCad/Housings_SOT-23_SOT-143_TSOT-6.pretty. This last one contain has the footprint with the correct values.

ricardocrudo, 2015-04-22:
@michal777 @CarlPoirier, which one should be removed?

stambaughw, 2015-04-22:
That's fine but we should probably remove the incorrect footprints for
the Diodes_SMD library.

stambaughw, 2015-04-22:
Diode_SOT23 and Diode_SOT23_Handsoldering should both be removed from
the Diodes_SMD library.

CarlPoirier, 2015-04-22:
Thanks Wayne for reporting this. Let's remove the footprints from
Diodes_SMD and keep the ones from Housings_SOT-23_SOT-143_TSOT-6.pretty.

CarlPoirier, 2015-04-22:
Creating a _HandSoldering version from the correct footprint would be great
however. Else we lose that footprint.

Ricardo, I will approve your pull requests tonight.

ricardocrudo, 2015-04-22:
@CarlPoirier Housings_SOT-23_SOT-143_TSOT-6.pretty has a _HandSoldering version to SOT-23 as well.

CarlPoirier, 2015-04-22:
Indeed, haha! All is well, then.

CarlPoirier, 2015-04-24:
The duplicates have been removed.

closed 2015-04-24 by CarlPoirier