KiCad/kicad-library #186: Do you accepts 3d model files other than wings and wrl?

hyOzd, 2015-04-30:
Do you accept Freecad or Blender as source format for 3d models? I have a couple models that I would like to contribute.

Also do you accept X3D instead of wrl? Asking this, because blender cannot export to wrl format and it seems kicad works fine with X3D.

michal777, 2015-04-30:
Some .wrl exported from blender can crash KiCad (older builds are less robust) so I think it's safer to import 3ds or obj from blender to wings and export to wrl.

hyOzd, 2015-04-30:
What about freecad? Since freecad is a parametric modeler, it can be useful to have a freecad source file around, I think. 

Though, I must add; freecad's exporter features seems to be unstable. A couple days ago, I wasn't able to export a simple cylindrical shape to wrl. Maybe to play it safe, I should contribute Freecad+wings+wrl? But there is no clear path from freecad to wings. Only method I could find atm is this: Freecad export WRL -> Blender import WRL -> Blender export 3DS/OBJ -> Wings import 3DS/OBJ. Maybe I should forget about freecad :)

michal777, 2015-04-30:
Doesn't freecad export 3ds/obj that can be opened in wings? If it doesn't exporting freecad->blender->wing worked ok for me.

hyOzd, 2015-04-30:
Freecad can export OBJ. But color information is lost and some normals are wrong. Not really useful.

cbernardo, 2015-05-01:
Not to mention you may not have colors, and you definitely
won't have colors if you go from FreeCAD to wings then VRML. 
There are my old VRML modeling tools which can be scripted:

http://kicad3dmodels.sourceforge.net/
  

 For very complex models it can take a lot of time to make
a good script, but I think for most models it's much quicker
than Wings3D (and much better) and maybe about the same as
FreeCAD in the time it takes to make a model.

- Cirilo

hyOzd, 2015-05-01:
Here is another method I use to move from freecad to wings.

- freecad: Export different colored parts as different mesh files (STL)
- blender: import mesh files
- blender: apply colors to different parts
- blender: export to X3D/3DS
- wings: import from 3DS

I use this method when freecad wrl exporter messes up. I think blender can be skipped if you know how to use wings.

But my original question was if I should add freecad files to repository?

s-light, 2015-05-03:
Thanks for asking this question hyOzd!
i also do most of my part-3D modells in Blender and or FreeCad.
i have written down a [workflow at the kicad.info forum] (https://forum.kicad.info/t/my-1st-3d-model-for-kicad/547/2?u=s_light) - this was written some time ago - so eventually with newer KiCad / FreeCad / Blender Versions things can be different...

i would like to see other source files in the KiCad repository..

hyOzd, 2015-05-03:
I think best part of using freecad is to be able to create parametric/scripted parts very easily.

[Here](https://dl.dropboxusercontent.com/u/62663835/smd_pin_header.fcstd) I have created a SMD Pin header model. And I have created a very simple macro-python script to change the number of pins. As you can see below; script will find array objects that I have labeled with a specific keyword and update the "repeat number". That's it. It's just a couple more lines to be able to generate whole range of 1-40 pin headers.

```python
import FreeCAD

doc = FreeCAD.activeDocument()

arrays = [a for a in doc.Objects if "paramX" in a.Label]

# change paramater
pX = 12

for a in arrays:
    a.NumberX = pX

# finalize
doc.recompute()
```

This version of the script should be run when FreeCAD document is open, from the macro dialog. It's also possible to code this as a standalone script, that will open the document, make the changes then save or export to a mesh format. But this may limit some functionality, such as accessing the color/material information of the object. Also there are some other non-destructive problems, such as object colors are not displayed correctly and visibility status of some the objects changes.

Currently I'm trying to figure out a way to reliably export FreeCAD objects to X3D. FreeCAD's VRML export is really messed up. It's exporting same shape multiple times including edges and points as separate objects. Don't know why.

s-light, 2015-05-04:
>  It's exporting same shape multiple times including edges and points as separate objects. Don't know why.

i noticed this behavior when i imported the FreeCad-exported files in blender..
if i remember correctly it consists of:
for every object:
--> as 'PointCloud'
--> as 'EdgeCloud'
--> as 'Surface'
and all three variants two times... (with no visible differences in Blender)
i have asked in the FreeCAD forum about this: http://forum.freecadweb.org/viewtopic.php?f=3&t=10815 next would be to ask also in the blender forum but i think one thing after the other..

Fat-Zer, 2015-05-17:
> Currently I'm trying to figure out a way to reliably export FreeCAD objects to X3D. FreeCAD's VRML export is really messed up. It's exporting same shape multiple times including edges and points as separate objects. Don't know why.

I've also run into that just yesterday making a [led display model](https://github.com/Fat-Zer/kicad-library/raw/CX56-12-3d/modules/packages3d/Displays_7-Segment.3dshapes/Cx56-12.fcstd) and advice on the FreeCAD forum seems to work: then copying everything to another document you get the clean wrl which is being read and displayed fine by the recent bzr kicad build.

CarlPoirier, 2015-05-17:
Would any of you volunteer for writing a simple step by step tutorial about
the process? We could put in in the wiki on Github.

Fat-Zer, 2015-05-17:
They just commited a fix to FreeCAD vrml export module so now it works more straight forward... So now it seems it's enough just to export to wrl.

I've tried to produce some broken output that kicad won't handle correctly but I couldn't =) ... 
On the other hand I'm not a vrml guru so I can't evaluate how much the output confirms to standards (or the kicad implementation either)...

CarlPoirier, 2015-05-25:
Woah that's nice to know!

michal777, 2015-06-03:
I don't know if wrl from freecad can be accepted, I think it will be better to ask developers
https://answers.launchpad.net/kicad

easyw, 2015-07-15:
Hi to all,

I've seen you model your 3D components in FreeCAD... That's great! :)
With the new daily build version of kicad it is possible to use directly VRML modules generated by FreeCAD and that would be available also in the up-coming new stable release. 
I have done a script that takes the FreeCAD ability to export VRML and STEP formats to build a STEP version of the board with models, perfectly identical in displaying in kicad 3d-viewer and in FreeCAD or MCAD software. This should make enclosure design easier...
I called the script 'kicad StepUp' and it is available at sourceforge.net and launchpad.
Here it is possible to see the script in action:
http://youtu.be/Ukd47VXYzQU
a demo board with some models is also downloadable
[image]

I would like to make a repository of models created in FreeCAD with the corresponding STEP and VRML model...
If you would like to share your models, I'll be happy to add these to the 3D repo!!! :)

thank you for any suggestion and feedback,
Maurice

ricardocrudo, 2015-08-02:
Quote from @easyw 

> Hi, would be possible to add also the STEP model exported from FreeCAD file Cx56-12.fcstd?
https://github.com/easyw/kicad-3d-models-in-freecad/raw/master/misc/Cx56-12.step
the STEP model is useful to be used with kicad stepup script
This script is aimed to give kicad the ability to have an identical view of board and modules in kicad 3d-viewer and in FreeCAD or MCAD software
it can be found from http://www.kicad-pcb.org/display/KICAD/External+Tools, kicadstepup
https://sourceforge.net/projects/kicadstepup/

Does someone comment on this? Maybe @CarlPoirier

easyw, 2015-08-03:
@ricardocrudo  thank you for posting it
@CarlPoirier 
just few words to explain why I've done the kicad stepup script and why I'm asking to add FreeCAD model to the repo:
When you need to give your 3d artwork to MCAD people you have to use step as interchange format.
Any other format would not be accepted.
Before the script the only way I know to export the pcb with models was to export vrml and then convert it to step; but any model converted from vrml would be very un-optimal and not very useful for mechanical design... the conversion, using whatever sw you find, will generate a bunch of triangles.
If you search in the mailing list the problem is quite heavily discussed.
That's why I developed the script; you can design your pcb and check everything in kicad 3d-viewer and then export the model in step to use and share your mechanical design with the same appearance/dimensions in both environments.
To test the script in real world, I used hackrf-one board and I generated all 3d vrml modules to be used in kicad, directly from step modules downloaded from the manufacturers or from 3dcontent central library. All the modules were converted to wrl with freecad and displayed in kicad without any error or crash; so I would say that FreeCAD could be used as second source to model 3D object to kicad, side by side with wings3d. Moreover FreeCAD can create parametric model, with real dimensions as per the physical objects.
here there is a part of the discussion at the developers mailing list where the script has been posted:
http://www.mail-archive.com/kicad-developers@lists.launchpad.net/msg14131.html
the screenshots of the hackrf-one have been included at the kicad home page
http://www.kicad-pcb.org/display/KICAD/KiCad+EDA+Software+Suite?preview=/589828/16973826/hackrf-one-stepup-kicad-3d-viewer.png
http://www.kicad-pcb.org/display/KICAD/KiCad+EDA+Software+Suite?preview=/589828/16973825/hackrf-one-stepup-catia-assembly.png
Adding FreeCAD models (and eventually also STEP models) to the repo will help users to make enclosure design easier.
Regards
Maurice

CarlPoirier, 2015-08-04:
Hi,

This is very interesting. Would there be a way (some script?) to migrate all existing 3D models from KiCad to STEP automatically? I haven't played around with 3D modeling and KiCad so I'm not sure what is doable.

easyw, 2015-08-05:
Hi Carl,
unfortunately it is not possible to convert from Vrml to STEP nor from wings3D to STEP to obtain a usable MCAD model.
I know there is a huge amount of 3d models done with wings3d, but I would suggest to accept also model done in FreeCAD.
FreeCAD can export to VRML in a fully compatible format to be read by kicad.
FreeCAD can import STEP models and export to VRML.
FreeCAD can create parametric MCAD models to be fully aligned to the component datasheet

thank you
Maurice

CarlPoirier, 2015-08-05:
Do you think we can populate at least most component 3D models easily by hand?

cbernardo, 2015-08-06:
I think 3D models will remain a challenge; no doubt there are manypetabytes of models out there. There are many companies who willsell you access to their collection. However, more recently a numberof manufacturers have been providing 3D models in a number offormats which almost always include IGES and STEP.
Due to the huge number of models I think individual users should beresponsible for their own IGES or STEP models. We can provide a fewof the more common models though such as a variety of SMT resistors,SOT-xx DPAK-xx SOIC, QFN, DFN, TQFP. Since FreeCAD has fixed itsVRML export enough to be useful to KiCad, I think Maurice's sugestionto create VRML models from STEP or FreeCAD models is a good idea.
- Cirilo

easyw, 2015-08-06:
@CarlPoirier 
Hi, I'm not very used to FreeCAD, but I think that modeling 3D components in kicad would be easier than in Wings3D. I see there is sketcher, extruder, chamfer, fillet etc...
Moreover I see that @hyOzd has done some script to create parametric/scripted parts
here there is one of his models
https://dl.dropboxusercontent.com/u/62663835/smd_pin_header.fcstd
may be if he could be involved in the process, we could have a scripted/parametric library of 3D models.
I think that parametric way to produce models would be the answer to have a fully populated lib in short time...
Maybe if @hyOzd would model e.g. a SOIC8 model, with a parametric script it could be possible to have all the SOIC SSOP QFP SOT (Gullwings in general) 3D models in a flash!
That process would be replicated for Chip resistors, Chip capacitors, QFN, Tantalum capacitors etc.
Thank you
Maurice

hyOzd, 2015-08-06:
@easyw I was going to talk about this soon. Thanks for calling me out :) 

Creating parametric files in FreeCAD, at least the way I did has some problems. First let me describe you my method:

1. I draw a part using FreeCAD, I use parametric features as much as possible. Such as 'array' tool from "Draft" workbench
2. I label some of those features (such as array tool) with a special syntax
3. Using a small script, I change the parameters of those specifically labelled features to my needs. For example I draw a 4 pin header leveraging the array tool, then I use a script to change all '4's to '8', and now I have a 8 pin header.

What's the problem? Well, FreeCAD is not smart enough. For example if I try to apply fillet to the top face of a cube, after I updated the size of cube, fillet may be broken. When the shape changes, FreeCAD cannot determine the correct face to apply the fillet. It doesn't have all the features to be a truly parametric modeller.

I see a couple solutions to this; 

1. use only the tools that can be altered without side effects, avoid tools such as fillet
2. use FreeCAD's Python API to draw the shape from scratch. Can be hard to design a model this way, and it's harder to maintain
3. do the first option first, draw without problematic tools, then use pyhon APIs to add extra detail. I just don't like this idea. It's too complicated.

----

But I would like to present to you another alternative: [cadquery](https://github.com/dcowden/cadquery/)

> CadQuery is an intuitive, easy-to-use python based language for building parametric 3D CAD models

If you have heard of OpenSCAD, this is a similar tool. It's just that this one is a python library and has a syntax inspired by jQuery. At the moment, cadquery uses FreeCAD as its underlying 3D modelling engine. "In theory" it can do anything that FreeCAD is capable of. 

I must say, designing a detailed part using cadquery is not very easy compared to a visual tool. But if you compare it to FreeCAD's python API it's a much saner option. 

[Here](https://bitbucket.org/hyOzd/freecad-macros/src/e85defa66322bad7f2ef5df6fac17b5e94268eb8/make_pinheaders.py?at=master) is a cadquery script that I have created to model all the pin headers ranging from 1 to 40 pins. You want 80 pins? No problem. It even uses parallel processes to reduce build time. Note that my script generates X3D mesh files. Cadquery supports STEP export, but I haven't tried that yet. I will try that as soon as possible.

I have more examples; [a qfp part generator](https://bitbucket.org/hyOzd/freecad-macros/src/e85defa66322bad7f2ef5df6fac17b5e94268eb8/make_qfp.py?at=master), [a DIP part generator](https://bitbucket.org/hyOzd/freecad-macros/src/e85defa66322bad7f2ef5df6fac17b5e94268eb8/make_dip.py?at=master) .

If you want to see the end results of these scripts right now, you can look at these blender files. Note that some of them are modified by hand just a little bit;

* [pinheaders.blend](https://bitbucket.org/hyOzd/pcb2blender-models/src/6f11d961ec0006f09d925a60d02bbf74aeb5093a/pinheaders.blend?at=master)
* [dip.blend](https://bitbucket.org/hyOzd/pcb2blender-models/src/6f11d961ec0006f09d925a60d02bbf74aeb5093a/dip.blend?at=master)
* [qfp.blend](https://bitbucket.org/hyOzd/pcb2blender-models/src/6f11d961ec0006f09d925a60d02bbf74aeb5093a/qfp.blend?at=master)

Some notes/disclaimers:
* Although creating cadquery scripts can be hard at first, reuse rate can be really high. Especially in the case of plastic IC packages. For example the code that I've used to draw the plastic part of the DIP and QFP models is almost the same. It's not the same, just because I was lazy.
* cadquery is an experimental/under development software. It uses FreeCAD as its engine which is also under heavy development itself. But then again, KiCAD is not very stable either :)
* I haven't tested the STEP export. I'm not sure if it can handle multicolored parts.
* Cadquery doesn't have capability to export X3D. I use [my own X3D export script](https://bitbucket.org/hyOzd/freecad-macros/src/e85defa66322bad7f2ef5df6fac17b5e94268eb8/export_x3d.py?at=master). This scripts generates a very pure output but it's not accepted by KiCAD although its valid X3D file. Can be easily fixed though.

I'm already on my way to creating more generator scripts. But I could use some help. Gathering the parameters for different variations, testing and making sure that all works without problems takes most of the time. I could use some help in that regards.

easyw, 2015-08-06:
@hyOzd Hi!
I see the problem in parametric modeling starting from a Draft manually made...

I think use FreeCAD's Python API to draw the shape from scratch would be very promising ...
adding parametric values to the script will build all the models
I know that could be quite complex at the begin, but it has strong potentials...
and also to see the macro in action, building the model should be nice :)
May be starting with chip capacitors would be the easiest...

I consider very promising also the cadquery option, providing you can export your model in STEP with colors...
then the model can be fused in single part and converted by FC to vrml, to be used in kicad, just with a simple external script
Unfortunately any other model (X3D or VRML) would not be useful for our mechanical purposes

Please let me know when you have exported some model to STEP, I will be happy to test them with the script
thank you
Maurice

easyw, 2015-08-06:
@hyOzd Hi again...
I've tried some scripting with cadquery and I think it would be the right choice...
If you can model e.g. chip capacitors (and I don't doubt as I've seen your qfp cadquery script) there are no problem in adding some trick to color parts and export them to STEP and to VRML (after scaling)

I've done a little cadquery script to test it ...
I modeled just a chip with body (no pins) following cadquery example and I add some code to add color and export the result in STEP and VRML
here the script
https://github.com/easyw/kicad-3d-models-in-freecad/raw/master/cadquery/cq-ex2.FCMacro
FreeCAD 0.15 and 'cadquery-freecad-module' v0.18 is required
(https://github.com/jmwright/cadquery-freecad-module/archive/v0.1.8.zip)
to run the script in Linux:
freecad cq-ex2.FCMacro
in windows:
C:\FreeCAD\bin\freecad cq-ex2.FCMacro
the 3D models obtained are aligned correctly to be used with the kicad stepup script, so to be used for MCAD exchange format / modeling and in kicad 3d-viewer
thank you
Maurice

easyw, 2015-08-09:
Hi @hyOzd @cbernardo @CarlPoirier 
when I have been able to test the parametric script done by hyOzd, I see that the generating and conversion to STEP was just a step ahead ... :)
I discovered that to test the script I had to download also the CadQuery fork from hyOzd
Then I realized that the models where already generated with colors...
So I developed a script to fuse with colors, convert to STEP and scale and convert to VRML so to be ready to be used in kicad and with kicad StepUp script...
Now we have a start ... a repository of all QFP chips, generated with a parametric script!
Please have a try at:
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator
and at the first model generated:
https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/generated_qfp/qfp64_10x10_p05.step
https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/generated_qfp/qfp64_10x10_p05.wrl
https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/generated_qfp/qfp64_10x10_p05.FCStd

thank you a lot @hyOzd for your great work!
Maurice

easyw, 2015-08-11:
Hi @hyOzd again! :)
I've worked a bit on the qfp script and now it is possible to generate all Gull Wings ICs by the parametric script!
Your qfp script was done so well that adding the SSOP, TSSOP, SOIC etc models has been easy ...
I'm still using my libs to export to STEP wth colors and to VRML for kicad, but I'll keep an eye on your cadquery work, to switch to fully cadquery philosophy.
I adapted also the DIP an pinheader scripts to obtain all the models in STEP with colors.
I had to change slightly something with some models to avoiding malformed geometries generated sometimes by FC fusion ...
I would suggest you to add yourself (if you are not already) to the kicad developers mailing list
https://lists.launchpad.net/kicad-developers/
https://launchpad.net/~kicad-developers
you can get the edge of kicad trends from there ...
I'm going to post these comments also there, to ask if they may consider to add FreeCAD wrl models to the official repo.
Anyway, I would say that CadQuery is a GREAT tool for modeling 3D objects for kicad!
Thanx again and keep in touch
Maurice
here a TSSOP model recently generated
[image]

easyw, 2015-11-29:
@CarlPoirier @ricardocrudo 
I have done a first 'STEP' in building a **3D MCAD repo** for kicad
The repo will contain 3D models designed preferably in FreeCAD, with STEP 1:1 and VRML 1:1/2.54  many of them will come from the parametric generator for FreeCAD that @hyOzd and me have done.
**Contributors are very welcome!**
The first 3D models are
       *Capacitors_Tantalum_SMD.3dshapes*
those **models have been done and named to be fully compatible with the kicad module library**
       *Capacitors_Tantalum_SMD.pretty*
I'm going to add more 3D models to help kicad users with a MCAD library to be able to export pcb and parts to mechanical environment, for ECAD/MCAD collaboration and enclosure design.
see the topic in the forum
https://forum.kicad.info/t/3d-new-library-for-mechanical-cad-exporting-and-enclosure-design/1763/
[image]

hyOzd, 2017-06-24:
I see that freecad files are accepted in this repository: https://github.com/KiCad/packages3D-source .

closed 2017-06-24 by hyOzd