KiCad/kicad-library #191: Create a KiCad library scripts repository

CarlPoirier, 2015-05-02:
Hi @michal777 and @ricardocrudo,

I think it would be great to have your scripts as part of the KiCad organization. What if we created a new repository for them?

Also, as a matter of uniformization and portability, I suggest we write all of them in Python. @michal777, I volunteer to rewrite your C scripts.

michal777, 2015-05-02:
Yes, it would be great to have some repository for scripts.

I think that it will be also useful if we collect files with dimensions and other parameters that could be used to create both footprints and 3d packages with scripts made by @cbernardo. If we follow our convention and try to respect IPC-7351C the design of some of footprints is well defined and it's waste of time to draw them by hand. It's also easier to update the footprints when convention changes.
I think we can also ask users for they scripts if they can do something useful (eg. scripts generating varistors from table).

Here are footprint wizards (kicad scripting):
https://github.com/KiCad/kicad-source-mirror/tree/master/pcbnew/scripting/plugins
I think that it can be relatively easy to make them compatible with KLC. Functions which calculates drawings of silkscreen and courtyard could be copied from my script: 
https://github.com/michal777/Packages_Generator/blob/master/script_qfp_sop_qfn_dfn/pack_generator.c
printf can be replaced with self.Line. Unfortunately I've no idea how to change value, reference, 3d, layers etc. I don't know python. Example of silkscreen:

```
    def SilksSOP(self, LINE_WIDTH, CRTYD_CLEARANCE, pads_count_y, pitch, pad_size_x, pad_size_y, x_pads_center, body_size_max_x, body_size_max_y):
        """
        Draw klc silks for sop
        """

        start_line_y = end_line_y = body_size_max_y / 2 + LINE_WIDTH / 2;
        if start_line_y <= (pitch * (pads_count_y - 1) / 2 + pad_size_y / 2 + 1.5 * LINE_WIDTH):
            start_line_y = end_line_y = pitch * (pads_count_y - 1) / 2 + pad_size_y / 2 + 1.5 * LINE_WIDTH;
            start_line_x = -x_pads_center / 2 + pad_size_x / 2;
            end_line_x = -start_line_x;
            self.Line(start_line_x, start_line_y, end_line_x, end_line_y)
            start_line_y = end_line_y = -start_line_y;
            start_line_x = -x_pads_center / 2 - pad_size_x / 2 - LINE_WIDTH / 2;
            self.Line(start_line_x, start_line_y, end_line_x, end_line_y)
        else:
            start_line_x = end_line_x = -body_size_max_x / 2 - LINE_WIDTH / 2;
            start_line_y = -body_size_max_y / 2 - LINE_WIDTH / 2;
            end_line_y = -pitch * (pads_count_y - 1) / 2 - pad_size_y / 2 - 1.5 * LINE_WIDTH;
            self.Line(start_line_x, start_line_y, end_line_x, end_line_y)
            start_line_x = end_line_x = -start_line_x;
            self.Line(start_line_x, start_line_y, end_line_x, end_line_y)
            start_line_y = -start_line_y;
            end_line_y = -end_line_y;
            self.Line(start_line_x, start_line_y, end_line_x, end_line_y)
            start_line_x = end_line_x = -start_line_x;
            self.Line(start_line_x, start_line_y, end_line_x, end_line_y)
            start_line_y = end_line_y = -body_size_max_y / 2 - LINE_WIDTH / 2;
            start_line_x = -body_size_max_x / 2 - LINE_WIDTH / 2;
            end_line_x = body_size_max_x / 2 + LINE_WIDTH / 2;
            self.Line(start_line_x, start_line_y, end_line_x, end_line_y)
            start_line_y = end_line_y = -start_line_y;
            self.Line(start_line_x, start_line_y, end_line_x, end_line_y)
            start_line_y = end_line_y = -pitch * (pads_count_y - 1) / 2 - pad_size_y / 2 - 1.5 * LINE_WIDTH;
            start_line_x = -body_size_max_x / 2 - LINE_WIDTH / 2;
            end_line_x = -x_pads_center / 2 - pad_size_x / 2;
            self.Line(start_line_x, start_line_y, end_line_x, end_line_y)
            
        start_line_x = end_line_x = -x_pads_center / 2 - pad_size_x / 2 - CRTYD_CLEARANCE;
        start_line_y = -body_size_max_y / 2 - CRTYD_CLEARANCE;
        end_line_y = body_size_max_y / 2 + CRTYD_CLEARANCE;
        self.Line2(start_line_x, start_line_y, end_line_x, end_line_y)
        start_line_x = end_line_x = -start_line_x;
        self.Line2(start_line_x, start_line_y, end_line_x, end_line_y)
        start_line_x = -x_pads_center / 2 - pad_size_x / 2 - CRTYD_CLEARANCE;
        end_line_x = x_pads_center / 2 + pad_size_x / 2 + CRTYD_CLEARANCE;
        start_line_y = end_line_y = -body_size_max_y / 2 - CRTYD_CLEARANCE;
        self.Line2(start_line_x, start_line_y, end_line_x, end_line_y)
        start_line_y = end_line_y = -start_line_y;
        self.Line2(start_line_x, start_line_y, end_line_x, end_line_y)

ricardocrudo, 2015-05-02:
That is really great. I had thinking to rewrite michal script in python, actually I already had started creating the class to store, in a friendly way, the parsed s-expr (kicad_pcb) file. Maybe I could keep going with this work creating the class and once I have it done I push it, after that, new scripts can be created using the class API. Same way that I did with [schlib](https://github.com/ricardocrudo/kicad-utils/tree/master/schlib) and [sch](https://github.com/ricardocrudo/kicad-utils/tree/master/sch). @CarlPoirier what do you think?

CarlPoirier, 2015-05-16:
Hi Ricardo, it sounds great. Go ahead and one we have that, we'll build on it.

I'm thinking of creating a new repo called kicad-library-utils.

ricardocrudo, 2015-05-17:
kicad-library-utils sounds great to me.

I already pushed a bunch of commits with kicad_mod class implementation. Today I'll push the first kicad_mod checker working with some rules. They can be used as example to future checking scripts. 

BTW, the rule 6.9 says: "Value and reference have a height of 1mm.". The @michal777 scripts check not only the height but also width and thickness, should the rule be updated or Michal did an extra checking? :)

CarlPoirier, 2015-05-17:
I have created the repo https://github.com/KiCad/kicad-library-utils.

Ricardo, do you want to put your name as the author at the top of your files before we commit them to it?

ricardocrudo, 2015-05-17:
Carl, If is OK for you I can just change my .git/config to point to this repo and push it bringing all git history. I don't care about my in the files, the git history has too much more to say :)

michal777, 2015-05-17:
The thickness in in other rule - about line width. The width - I forgot it's not in KLC, maybe should be (I think the idea was to have one standard, predictable sizes).

CarlPoirier, 2015-05-18:
@ricardocrudo Yeah that's fine.

As for the value thickness, it's because it once was on silkscreen and thus was regulated by 6.5. I will indicate it explicitly in the KLC.

ricardocrudo, 2015-05-18:
I pushed the code into kicad-library-utils. I did some of the checking that Michal does in his script but still missing many other. Unfortunately, I did not spend too much time doing a proper documentation about the kicad_mod API/class, but I hope the checkrule files be useful as start point to new scripts.

This is a working in progress and will getting better with the time. I know there are a lot of things that can be improved and new ones created, since small help text adjustments until create scripts to rotate footprints or create automatically the courtyard line. Would be nice if we use the github to register this.

I know the README is very poor and not useful. Would be great make a clean up putting aside the non useful information for the regular users, a requirements list (python for example) and a good step by step guide with practical examples.

Finally, I've to say that I'm a Linux user and despite the code is in python I expect to hear if the scripts are running fine in the other OSes.

andres53016, 2015-05-19:
It sound great !!!
I'll make something usable to generate footprints from tables.

ricardocrudo, 2015-07-25:
Done.

closed 2015-07-25 by ricardocrudo