KiCad/kicad-library #194: DO-214AB 3D model scaling and orientation issue.

stambaughw, 2015-05-04:
The 3D model Diodes_SMD.3dshapes/Diode-DO-214AB.wrl is not scaled or oriented correctly (see image).  The scaling is off by 1/2.54 and the cathode band is rotated 80 degrees.

[image]

ricardocrudo, 2015-05-05:
It's ok to me.
[image]

stambaughw, 2015-05-05:
> kicad-test

Do you have the 3D model scaling set to 1 in the footprint 3D model
settings?  I had to set it to 1/2.54 and rotate the Z axis 180 degrees
to get the model to fit the footprint.  I just did a git pull yesterday
so I'm assuming I have the latest version of the 3D model.

ricardocrudo, 2015-05-05:
No. scale set to 0.39 and z zaxis set to 180. Like here: https://github.com/KiCad/Diodes_SMD.pretty/blob/master/Diode-DO-214AB.kicad_mod

stambaughw, 2015-05-05:
I guess this is OK in the short term.  My preference is to have our
models scaled and oriented to our footprints.  I'm sure there are other
users with legacy DO-214AB footprints who will get surprised like I did.

michal777, 2015-05-05:
There are much more 3d models drawn in mm and scaled in kicad_mod file and it's not easy to scale all the wings and wrl.

stambaughw, 2015-05-05:
It would be nice if all of the 3D models had the proper scaling and
orientation but I guess we can live with it as long as the footprint 3D
settings are correct.

cbernardo, 2015-05-05:
>________________________________
> From: michal777 <notifications@github.com>
>To: KiCad/kicad-library <kicad-library@noreply.github.com> 
>Sent: Wednesday, May 6, 2015 12:26 AM
>Subject: Re: [kicad-library] DO-214AB 3D model scaling and orientation issue. (#194)
> 
>
>
>There are much more 3d models drawn in mm and scaled in kicad_mod file and it's not easy to scale all the wings and wrl.

In those cases there's a simple tool in my 3D model toolkit that can scale
a VRML model. However, the tool may need to be updated to handle VRML1.0
models which are still the most common VRML format created by MCADs. The
tool is also very primitive and designed to work with models accepted by
KiCad's VRML engine ~2 years ago, so it will work with Wings3D output but
it hasn't been tested with many models from other sources. Of course there
is the nuisance that the model maintainers have to set up another tool
and run a script to adjust some models.

- Cirilo

michal777, 2015-05-05:
Approximate scale of the problem (libraries sorted by number of footprints with "scale (xyz 0.39"):

[image]

michal777, 2015-05-05:
BTW why do we use inches? Will the scaling 0.1mils/unit stay forever?

cbernardo, 2015-05-05:
>________________________________
> From: michal777 <notifications@github.com>
>To: KiCad/kicad-library <kicad-library@noreply.github.com> 
>Cc: cbernardo <cirilo_bernardo@yahoo.com> 
>Sent: Wednesday, May 6, 2015 8:35 AM
>Subject: Re: [kicad-library] DO-214AB 3D model scaling and orientation issue. (#194)
> 
>
>
>BTW why do we use inches? Will the scaling 0.1mils/unit stay forever?

This is just history, but unfortunately changing this will break
everything. The real problem is that VRML has a "unit" and officially
it is meant to be 1m. What really happens is that different people/
software use data ranges convenient for their specific application -
this means that models in the wild have many scale factors and (a)
we have no control of this and (b) there is nothing in the VRML
specification which can tell us what scale is used. Even the VRML
exporter will produce output in 3 different scales and unfortunately
I didn't include a scale equal to the scale used by KiCad. (This
is not a problem though - I can eliminate the '1m' scale and use
'0.1in' instead.)

Anyway, I think 100mil/unit will indeed stay forever. These are just
numbers after all and as I said we need to scale models differently
depending on their source, so there is no good technical reason to
change so much code and change all existing VRML models just to
change the model unit.

- Cirilo

Shackmeister, 2017-06-12:
@stambaughw a properly scaled and aligned wrl file (with material properties) has been added in the repo, could you please consider closing this issue?

Shackmeister, 2017-07-01:
In case anybody is interested you can find the new model (step and WRL 2.0) PR here https://github.com/KiCad/packages3D/pull/49
Ill close this issue now, let me know if it should be reopened

closed 2017-07-01 by Shackmeister