KiCad/kicad-library #196: Resitor and capacitor symbols changed?

ondra6ak, 2015-05-08:
I've opend KiCad after a while of not using it and i've discovered, that symbols for resistor and capacitor are 1.27 mm shorter, therefore my schematic is borken...
Is there any explanation to that?

keruseykaryu, 2015-05-08:
=> https://github.com/KiCad/kicad-library/pull/104

ondra6ak, 2015-05-12:
OK
Why are there shorted symbols (compared to the rest of regular sized symbols) for resistor and unpolarized capacitor and at the same time R_small and C_small?

CarlPoirier, 2015-05-16:
Can you be more clear? Which parts are you talking about?

ondra6ak, 2015-05-20:
[image]

This screenshot shows it clearly, i think.
C11 is shorter then the rest of regular sized symbols.
It is the same with regular R symbol.

samdolt, 2015-05-20:
Currently, we have two set of basic symbol, with two different length:

- Normal size ( R / CP / C / D )

- Small size ( R_Small / CP_Small / C_Small / D_Small )

This was a choice made in https://github.com/KiCad/kicad-library/pull/104 <https://github.com/KiCad/kicad-library/pull/104>

I think it’s useful to have this two size, because if you start a small project, bigger looks better. But in a project with
a lot of symbol in one sheet, smaller could be better.

ondra6ak, 2015-05-23:
Oh, it's problem with KiCad loading symbols from cache.
I agree that the symbols are now better.
Thank you.

CarlPoirier, 2015-05-25:
Glad to know you found the issue.

Fat-Zer, 2015-05-26:
By the word, potentiometer symbol is left unchanged...

CarlPoirier, 2015-06-06:
Indeed! I do think it could be made to match the normal size resistor.

closed 2015-07-25 by ricardocrudo