KiCad/kicad-library #197: Opto lib component LTV-354T may have wrong pin assignments

lsearl-thumpdataworks, 2015-05-08:
Looking at the datasheet
http://optoelectronics.liteon.com/upload/download/DS70-2001-004/S_110_LTV-354T%2020140520.pdf
I believe that the pin numbers should be 1, 2, 3, 4 instead of 1, 3, 4, 6 since the footprint is an SMD-4.

keruseykaryu, 2015-05-09:
Its depending on which housing will be used. The datasheet does not clearly give us an information about the housing name. The "Mini-flat package: 2.0mm profile" is not enough. I'm not sure if we have to call this "Mini-Flat-4", "MFSOP-4", "2.54SOP-4" or anything else...

I'm sure that we can not use SMD-4 for this symbol. The "SMD-4" is a modified DIP-4 with "gull-wing lead" (for example http://eewiki.ru/wiki/SMD-4).

During design of this part I assume that this optocoupler has modified MFSOP-6 housing. Using this footprint everything is fine with this pin-out. And it is compatible with its equivalents from TOSHIBA.

lsearl-thumpdataworks, 2015-05-09:
I looked at this again. You are correct. It will work with the MFSOP-6
footprint with the current pin assignments. I apologize.

leon

keruseykaryu, 2015-05-09:
I understand your doubts. I had it too.

The LiteOn on his website group these optocouplers as SOP-4 (Mini-Flat). If we/sb decide to create separate housing for this, we can adjust pinning in these symbols as well.

closed 2015-05-31 by keruseykaryu