KiCad/kicad-library #200: Some common schematics for OpAmps

Fat-Zer, 2015-05-14:
[image]

Now the library contains a handful of different OpAmps images, some of them are not conform to current conventions. I understand that most of them exist by historical reasons, but IMHO now it looks like a huge mess... 

I don't know what will be the best solution, but the most straightforward way I can think of is the slightly update the library format, some my ideas on it was listed [on the maillist on january]( https://lists.launchpad.net/kicad-developers/msg16325.html). On the other hand this is the most tough way and it can break compatibility of older versions of kicad with newer libraries.

So I at least propose a minimum solution: to choose a "best" OpAmp image that conforms to the current conventions (may be include it as some generic device) and recommend to use it for new devices and may be update some older devices to new image.

How does it sounds for you?

ricardocrudo, 2015-05-14:
I think would be great have, always that possible, the same shape. But, I like of the idea to make it at once, and to minimize the troubles that this could raise, make an announcement to the users, as was done to diodes A/K change, could be a way. In this, any symbol that someone pick to start a new OpAmp would be OK. In short, I'm in favor to make the KiCad library KLC compliant as fast as possible.

CarlPoirier, 2015-05-16:
I vote for uniforming this as well.

Fat-Zer, 2015-05-21:
[image]
Here is schematic I propose. can be vied here: https://github.com/Fat-Zer/kicad-opAmp-test
Please review it before I start convert the library (unless there will be some other volunteers).
 - OpAmp — a general OpAmp. (pin numbers are random)
 - Comp — a single power only OpAmp with V- as Gnd.
 - OpAmp-x2 — a dual opAmp
 - OpAmp-x2-v2 — a second variant of dual OpAmp, I like it more. if there will be no objections I'll use it.
 - LM158 — a preview of real schematic of a dual OpAmp (nothing special in that case)
 - CA3130 ­— a preview of another real opAmp with additional outputs.
 - CA3130-v2 ­— the second version of the same. pros: have place for output names. cons: looks a bit worser IMO.

So... I'm looking forward to hear your opinion at least on the complicated opAmps schematic...

ricardocrudo, 2015-06-14:
I like them, except CA3130-v2.

tahiro86j, 2016-01-02:
So, it looks like I'm asking the (essentially) same question at #354 with hesitations of being a beginner in GitHub.

I would vote for this be done.

Now, most modern OpAmps/Dual-OpAmps/Quad-OpAmps indeed have near-identical pin-layouts, but I feel that the common-symbol to be created should mention "which type it follows". I cannot really name the very first OpAmp to have pin-layout of, say NE5532, but that product indeed was the rule-setter for this (I think it was one of the Raytheon products).

Like mentioned, avoiding confusions could be a challenge, I agree.

tahiro86j, 2016-01-02:
The commonly-known layout for dual-OpAmp in DIP-8 seems to have been introduced by Raytheon's RC4558 in 1970s'. Successors just copied the same pin-layout. (see: http://www.electronics.dit.ie/staff/mtully/741%20tutorial/741%20Op-Amp%20Tutorial.htm)
So, why not name it something like RC4558-style?

tahiro86j, 2016-01-03:
Also, to be addressed is the treatment of power-input pins for multi-OpAmp ICs. This, I am saying on behalf of this user who felt the same about this issue.
(Expressed and translated in https://twitter.com/tahiro86j/status/683504602059177984 ).

I once tried to use "TL072" in Eeschema and there I placed both A-unit and B-unit on the schematic. Because I didn't know they both had to be connected to the same power-line (instead of marking one of them as NC), there was no way but for me to also strongly feel that this should be fixed someday.

Could this also be considered. please?

CarlPoirier, 2016-01-17:
@Fat-Zer How is this going? Are you still planning on finishing this?

Fat-Zer, 2016-02-20:
Sorry, I haven't showed up for a while it reveals that most of work was done a while ago by Oliver Walters long time ago. Great thanks to him. =)
Currently there are only four slightly different images: CA3130, AD620, LM733CH and LM386.
Others got some smaller variations too like text field positions and outline width.

I'd like to fix the four currently different now... Don't know if I should touch others.

closed 2017-09-15 by SchrodingersGat