KiCad/kicad-library #204: Problem with pin name and numbers in some components of regul library

Calu85, 2015-05-15:
Using component 78L05 from regul library and footprint Housings_TO-92:TO-92_Inline_Narrow_Oval produces an error when pcbnew reads netlist. The error message is:
```
Error: Component 'U1' pad 'GND' not found in footprint 'Housings_TO-92:TO-92 Inline Narrow Oval
Error: Component 'U1' pad 'VI' not found in footprint 'Housings_TO-92:TO-92 Inline Narrow Oval
Error: Component 'U1' pad 'VO' not found in footprint 'Housings_TO-92:TO-92 Inline Narrow Oval
```
I think the problem is in the schematic component. For example, pin table of 78L05 shows both number and name fields have the same value, namely [VI, VI], [VO,VO] ,[GND,GND], instead of [1,VO],[2,GND],[3,VI].
I modified the schematic component to have number 1,2,3 in the numbers fields and it solved the problem.
Is this a problem in the library? I have searched a little and I have found some people with the same problem in older versions of kicad.
I could fix it and make a pull request, but first I'd prefer to be sure that this is a problem and not something I am doing wrong in pcbnew.
Many thanks.

keruseykaryu, 2015-05-15:
Yes, this is an issue that should be fixed. However I see the two different ways to solve this issue:

1. We fix the pin numbers for generic symbols of these linear regulators and leave them in `regul` library for _quite backward compatibility_. Of course this will not be the full compatibility, because we need to apply __KiCad Lib Convention__ (KLC) rules. So pin endings might/will be not at the same positions anymore.
2. We remove all these generic symbols, because we have many counterparts of 78L05 series regulators in this library: LM78L05ACZ, MC78L05AACP, which have correct pinning, link to datasheet, housing and are correct in terms of KLC, and might be used as template symbols for others.

I'm rather for this __second option__, even if this smash a few schematics (_We did already worse things_ :) ).

That is my position on this issue. Now I'm waiting for Yours, and other key librarians: Carl, Ricardo, Michał (_Sorry if I skipped someone_).

CarlPoirier, 2015-05-16:
I'd go for deleting them as well.

keruseykaryu, 2015-06-25:
The pins names issues will be fixed when https://github.com/KiCad/kicad-library/pull/223 pull request will be merged. The question about remove generic symbols still remains...

ricardocrudo, 2015-06-25:
I can take advantage and remove this symbols too. They are: 7805, 78L05, 7805MS, 7905, 79L05.
Any other else?

Removing them I'll have to convert some LM7*** alias to real symbols.

Shackmeister, 2017-07-01:
@Calu85 Hi can you please consider closing this issue if its not relevant any more?

Calu85, 2017-07-02:
OK. I hadn't noticed it was still opened.

closed 2017-07-02 by Calu85

gechxx, 2017-09-19:
Hi so what the resolution.Im still getting stuck on check not accepting the 7805

poeschlr, 2017-09-19:
You might want to give more details here. I have no clue what you are talking about.

jkriege2, 2017-09-20:
Please also note that regul.lib received a complete workover (see  https://github.com/KiCad/kicad-library/pull/948) Maybe that fixes your problems (i.e. update your libs and try to exchange your symbols for new ones ...)

Best,
JAN