KiCad/kicad-library #206: Encoder

Fat-Zer, 2015-05-17:
I want to add encoder to the component library. And I have some questions:

 - What image to use? 
I've prepared some variants in the test project: https://github.com/Fat-Zer/kicad-encoder-test
[image]
Notes about the variants:
5 — from walter's library, see #135. Got some problems: pins are placed on 50mil and have uncommon sizes
3 — an attempt to fix 5 according to the KLC
4 — have pins ends placed on a 50mil grid nodes along X axis.

 - What should be the pin order? ABC or ACB?
The ABC is more straightforward in the schematic sense, but most of real encoders have ACB pin order.

Watching forward to hear your opinion on this issue...

ricardocrudo, 2015-05-17:
These days I did an encoder too. This one: https://www.bourns.com/data/global/pdfs/PEC11R.pdf
I did the version with switch, it looks like this: 
[image]

ricardocrudo, 2015-05-19:
> What should be the pin order? ABC or ACB?
The C stands for common and it is the pin 2 in the encoders devices, thus the right is ACB.

ricardocrudo, 2015-05-19:
Taking advantage of this question, which library sounds better to save the this rotary encoders? sensors? And what about the pretty repo, I think there aren't one which fits good to them. What do you think?

Fat-Zer, 2015-05-19:
>The C stands for common and it is the pin 2 in the encoders devices, thus the right is ACB.

Mostly but not always. e.g. [PEC16 encoder](http://www.bourns.com/data/global/pdfs/pec16.pdf) have ABC pin layout. So as a generic device I think we should have both variants in the schematic lib (as for transistors) and pins in footprints should be numbered as usual...

>Taking advantage of this question, which library sounds better to save the this rotary encoders? sensors? And what about the pretty repo, I think there aren't one which fits good to them. What do you think?
I was gonna to add schematic to device library as a quite generic input device... I'm not sure what footprint lib to use... If we could create at least 5 footprints on staging level it make sense to create a separate library.

ricardocrudo, 2015-05-20:
> Mostly but not always. e.g. PEC16 encoder have ABC pin layout. So as a generic device I think we should have both variants in the schematic lib (as for transistors) and pins in footprints should be numbered as usual...

hmm, I see, so, yes, the variants sounds a good way.

@Fat-Zer will you create symbols to encoders with switch as well? If so, how would you represent it in your symbols? I think would be great if we had an unique style, avoiding mutations as happened with OpAmps (#200).

Fat-Zer, 2015-05-21:
>@Fat-Zer will you create symbols to encoders with switch as well?
Sure... but I want to determine what will be the general schematic first, to not draw a key for every variant that will be dropped anyway. 

Also there will be also an option: to draw the key as a separate elements like in multi part component or as a part of an object.

SchrodingersGat, 2016-04-28:
@Fat-Zer did you end up making a symbol for the encoder? I was going to make one but did a quick search and found this discussion. Was there any conclusion here? I really like the symbol that @ricardocrudo  drew as above.

Fat-Zer, 2016-04-28:
nope, I haven't done anything about that since the time of discussion... you may continue this yourself if you wish...

SchrodingersGat, 2016-05-22:
for reference https://github.com/KiCad/kicad-library/pull/541

Fat-Zer, 2016-06-20:
Due to the PR is merged, I suppose the bug may be closed...

SchrodingersGat, 2016-06-20:
Check out https://github.com/KiCad/kicad-library/pull/575

The original PR (now closed ) had some issues so I've opened a new one.

Fat-Zer, 2016-06-20:
By the word, do you have in plans add a footprint for at least one encoder model?
/No pressure/ just wondering...

SchrodingersGat, 2016-06-20:
Just the symbol for the moment.

diggit, 2016-07-09:
#575 merged, only ACB variant with and without switch for now. Anyone can open PR with others.

SchrodingersGat, 2017-01-21:
Closing as this has been addressed.

closed 2017-01-21 by SchrodingersGat