KiCad/kicad-library #217: Wrong pinout for XC95144PQ100 ?

do9jhb, 2015-06-07:
I wanted to use the Symbol of XC95144PQ100 for a project.
But when I has compared the Symbol with the pinout mentioned in the Datasheet of this (http://www.xilinx.com/support/documentation/data_sheets/ds056.pdf), all Pins are on other Positions.
Is these desired or a bug?

keruseykaryu, 2015-06-07:
This is an old Xlinix library and might be buggy. Almost all KiCad libraries was designed and created by its users. So I don't know the intentions of the author of this symbol. I don't wanna guess.

I will mark this as bug, because the pins indeed doesn't match. If you can fix this issue, please do so and create pull request.
Because this is the CPLD series device, I hope any updates will not be in conflict with **Alex's** work **https://github.com/KiCad/kicad-library/pull/170**

ricardocrudo, 2015-07-24:
@do9jhb the right datasheet for XC95144PQ100 is: http://www.xilinx.com/support/documentation/data_sheets/ds067.pdf

Shackmeister, 2017-07-01:
@jbtronics Im not really sure if this relevant anymore, could you please consider closing this issue if not?

bobc, 2017-07-05:
The confusion here is that XC95144 is an old (obsolete) series, the new series is named XC95144XL, but Xilinx did not keep pin compatibility between the  old and new series.

Ideally, the obsolete versions should be tagged "obsolete - use XL for new designs" or something.

SchrodingersGat, 2017-07-05:
@jbtronics I know this is a long time since you raised this issue but would you be willing to contribute a fix? :)

bobc, 2017-07-05:
I believe there is nothing that really needs fixing here. The OP simply got confused by similar named but different parts. The "bug" label is incorrect, there is no bug. https://github.com/KiCad/kicad-library/pull/218 is incorrect and should NOT be applied.

I just did a partial check and the current symbols appear to be correct (XC95144PQ100 and XC95144XL-TQ100), if you look at the right datasheets.

SchrodingersGat, 2017-07-05:
Right, thanks for the investigation @bobc

closed 2017-09-15 by SchrodingersGat