KiCad/kicad-library #219: Issue in Gnd component

GauravSupal, 2015-06-11:
In new kicad version(2014 build), after creating netlist using net numbers, the gnd component is not assigning 0 node, it is considering any random value. Due to this I am facing problem in ngspice simulation. Can you please help me out.

astroelectronica, 2015-06-11:
Hi!

For simulation you must include pspice.lib and use 0 symbol for ground reference.

I think 0 symbol must be removed from pspice.lib and relocated to power.lib, but it just a personal opinion.

Remember to connect 0 line to pwr_flag symbol. 

Cheers!

fahim-oscad, 2015-06-12:
Hi,

I think there is change in kicad algorithm for ground component  as earlier ground was automatically taken as 0 but now the ground is taking random number.

Is there any way to fix the ground to 0 in current stable version ?

Shackmeister, 2017-07-19:
@gauravSupal is this still relevant? If not can you please consider closing this issue? :)

closed 2017-08-23 by Shackmeister