KiCad/kicad-library #279: Ask to join in KiCAD Libs contribution.

jbdwi, 2015-08-12:
Hi All,

I'd like to  join in libs contributtion @ KiCAD Project. While I'm still learning the contribution process / procedures ( unfortunatelly I felt I'm too slow to understand and follow the procedure ), I put my works on footprint and 3d models ( which I use in my project with KiCAD ) in my public-shared at Google Drive in folder KICADLIB_JBD.pretty at this address :

https://drive.google.com/drive/folders/0B3UxJ9sPS5F1flBUX2ZCSjBDR0NBbjdNaXlkZEVQbXBMZ1BjdVR4N3c4WmNZZG5lakdrNDg

To simplify, I put all the three related files ( xxx.kicad_mod, xxx.wrl and xxx.wings ) in one folder. 

Regularly I'll add the footprints and their related models ( wrl and wings ). The uploaded files are logged in file  "00 - UPLOAD FILES LIST - LOGS.ods". If available, some component's dimension / specs are included in subfolder DOCS.
 
Pls feel free to use them if you want. Pls free to modify, adjust and rename them. Advises, suggestions, corrections and criticism are very welcome. While I also monitor to this forum, you can  contact me  @ my email : jbdwiyono@gmail.com or inbox to my facebook account ( Josef B. Dwiyono , jb.dwiyono@facebook.com ). 

(Sorry for my bad english .. Hope you understand it ).

Thanks and best regards,
josef b dwiyono

jbdwi, 2015-08-12:
@ Carl Poitier, thanks for the guidance.

CarlPoirier, 2015-08-14:
Hi Josef,

I will have a look at it this weekend and also give you a hand with the procedure by email.

Carl

closed 2016-02-25 by CarlPoirier