KiCad/kicad-library #285: There's no convention for naming the "exposed pad" pin

cyrozap, 2015-08-19:
In ftdi.lib and microchip.lib it's GND, but in rfcom.lib it's GND_EP. The E-Pad isn't GND in every part, so it should probably have the signal name in it, but there should be a way to know it's not a normal pin without looking at the pin number. For this reason, I'm partial to something like \<signal\>_EP or \<signal\>_EPAD where \<signal\> is a placeholder for the signal name (GND, VSS, etc.).

CarlPoirier, 2015-08-30:
Why would you need to know in the schematics which signal is the exposed pad?

cyrozap, 2015-08-30:
It probably isn't strictly necessary, but it's handy to know in case you want to connect it to a separate net or something. In any case, I can't imagine a scenario where not knowing which pin is the E-pad would be better than knowing which one it is.

CarlPoirier, 2015-08-31:
The problem is that some part might have one package with an exposed pad and another package without. Then, you cannot name one pin *_EP because it will be named this way regardless of the footprint chosen.

jkriege2, 2016-11-24:
In several symbols, the EP-pin is called `PAD`or `PADGND`... or something in that style ...

SchrodingersGat, 2017-11-21:
@cyrozap  I am closing this due to long period of inactivity. The exposed pad numbering / naming follows the datasheet where appropriate.

closed 2017-11-21 by SchrodingersGat