KiCad/kicad-library #291: LED pin numbering disagreement

alexforencich, 2015-09-01:
The PCB footprints for LEDs have pin 1 = cathode.  The symbols have pin 1 = anode.  What gives?!?

ricardocrudo, 2015-09-02:
Do you have the last version of kicad-library?

alexforencich, 2015-09-03:
The damage has already been done.  I had to manually rebuild a handful of LED light towers as a workaround as the library was inconsistent when I laid out the boards.

alexforencich, 2015-09-03:
Yes, the library is still inconsistent.

ricardocrudo, 2015-09-03:
@alexforencich please let me know what exactly LED symbol and footprint you are talking about.

alexforencich, 2015-09-03:
Symbols: LED, LED2, LED_Small (library 'device)
Footprints: LED-10MM, LED-3MM, LED-5MM, LED-8MM, LED-0603, LED-0805, LED-1206 (library 'LEDs')

I suggest changing the pin numbers to A and K or A1, K1, A2, K2, etc. for both the symbols and the footprints for LEDs and diodes to remove any ambiguity.

ricardocrudo, 2015-09-09:
> I just updated to the latest git version and the library is still inconsistent. 

@alexforencich the led pin out is correct in device's library. pin 1 -> K, pin 2 -> A.

alexforencich, 2015-09-10:
Symbol pin 1 is the anode, not the cathode.  The problem is not that the footprint is incorrect, the problem is that the symbol and the footprint disagree with each other.  If I connect a wire to the anode pin of the symbol, I expect a net to be connected to the anode pin of the footprint when I add it to a board.  This is not the case as symbol pin 1 = anode while footprint pin 1 = cathode.

ricardocrudo, 2015-09-10:
@alexforencich checking here https://github.com/KiCad/kicad-library/blob/master/library/device.lib#L1063 led symbol has cathode on pin 1 and anode on pin 2. Can you point me where are you seeing this issue?

alexforencich, 2015-09-10:
Alright, so it seems that I have stumbled into a serious hole in the kicad library architecture.  It appears you swapped the pins on both the footprints and the symbols in commits https://github.com/KiCad/kicad-library/commit/9372e0c2706f78ecf2b104849f0cb703330b5385 and https://github.com/KiCad/LEDs.pretty/commit/c88801830f7701755de6ca48e3088b451b5ef703 .  This was a very bad idea due to the fact that kicad caches the symbols in the schematic.  It is also a bad idea in the case that somebody made a copy of the symbol library to make some modifications, which appears is what happened to me (I needed a 2 LED stack and a ferrite bead).  A schematic made with the pre-change symbols and then used with the post-change footprint will cause all of the devices to be hooked up backwards, without any sort of error messages or warnings.  The correct fix is an architectural one, moving to integrated footprint and symbol libraries like Eagle.  However, as much as that change should be made, it will unfortunately never be made.  So, the alternative would have to be convention and policy.  Policy would have to be no changes to pin ordering.  Convention would be to not use numbers for pins at all, but to name them with variations of A and C  - A and C; A1, C1, A2, C2; A1, A2, A3, C; etc.  With this change, if there is a version disagreement between the schematic and board, the pins will not be connected at all - i.e. there will be something obviously wrong.  Now, the footprint library may also get polluted with various similar footprints (RGB common anode and RGB common cathode), but without an independent mapping layer (like in eagle) you're either going to have duplicated footprints or duplicated symbols, both of which are less than ideal.

michal777, 2015-09-13:
Developers plan refactoring eeschema after the stable release so that the symbols will be embedded into schematic files.
According to the kicad library convention schematic symbols can be duplicated, footprints not.
Please always verify libraries, these are neither stable nor reliable, useful as starting point for your own. Be careful with kicad-library updates and "online libraries" (it's better to keep libraries on hdd or your own github account).
K=1, A=2 is common standard that we should follow.

closed 2015-09-13 by michal777

alexforencich, 2015-09-13:
Excellent.  If that is on the roadmap, then I will hold off on any kicad library development work (namely the comprehensive Xilinx library that I have been working on) until that feature is implemented in the development version.