KiCad/kicad-library #296: adding Connectors_JST_PH.pretty

pointhi, 2015-09-14:
I created all possible footprints for the JST PH connector (smt and tht, top and side variation).

Datasheet: http://www.jst-mfg.com/product/pdf/eng/ePH.pdf

The library is currently located inside my repository:
**https://github.com/pointhi/Connectors_JST_PH.pretty**

The footprints are created by a [self written python script](https://github.com/pointhi/kicad-footprint-generator) which can also be found in my repository (if something is made wrong or the Silkscreen has to be changed)

There are also 3D-models available: http://smisioto.no-ip.org/elettronica/kicad/kicad-en.htm
but they need some rework (placing) before we can add them

pointhi, 2015-09-20:
I also created footprints for the JST SH connector (top and side variation).

Datasheet: http://www.jst-mfg.com/product/pdf/eng/eSH.pdf

The library is also currently located inside my repository, and was created with my python script:
https://github.com/pointhi/Connectors_JST_SH.pretty

ricardocrudo, 2015-09-25:
I've talked with @pointhi in IRC, he is going to create PR's. @pointhi, please link here after created them.

pointhi, 2015-09-25:
Pull requests created:

* https://github.com/KiCad/Connectors_JST_PH.pretty/pull/1
* https://github.com/KiCad/Connectors_JST_SH.pretty/pull/1

closed 2015-09-25 by pointhi