KiCad/kicad-library #313: Resistor thru-hole vertical 3D model

imarin2, 2015-09-30:
Is there a reason why there is not such a wrl? I am doing one for myself, so I thinking if I should contribute it. Let me know.

cbernardo, 2015-09-30:
You mean like these?
http://sourceforge.net/projects/kicad3dmodels/files/models_v0.5/model_thru_resv_20121126.tar.bz2

imarin2, 2015-10-01:
Yes, precisely. I was not intending to have one model for each value (colour code), but definitely not complaining :). Thanks for pointing me to the source.

The question however still stands: Why are these not included in the repo? Are they being integrated or something like that?

cbernardo, 2015-10-01:
The devs didn't believe the VRML modeling tools belonged with KiCad
so they were never merged. Although old generated models are on SF,
the code is now hosted on github instead. I think the code should
just be packaged (though that would need volunteers) and people who
want to use the tools to generate models can simply install a
package and generate the models. The models aren't in the kicad
libraries on github simply because there are so many of them. Some

models have been put into the kicad libraries though such as the
qfp/qfn/soic models and I think the DIL models.

- Cirilo

imarin2, 2015-10-02:
Thank you very much for taking the time to reply. 

I now see my problem. I thought this GitHub repository was a space unlimited single source of libraries (symbol, footprint, 3D model) for KiCAD.

KiCAD is a great tool. However a tool is always as great as it takes you to get the job done. Libraries of electronic components (symbol, footprint, 3D model) have been such an important contributing factor in every EDA I have ever used, as they tend to suck an important part of the time one dedicates to a project. It is also a leading reason why a company using EDA A is reticent to switching to EDA B. Having an extensive library support plays quite a role in the switch.

I think that KiCAD, standing for what it is, opensource (and a great one), has an incredible potential in its community to get to an out-of-the-box vast library access (symbol, footprint, 3D model). However, in order to unleash that potential, storage space should not be a limiting factor in such an endeavour.

As many others, I also think WRML is not the most appropriate format for 3D modelling of PCBs as you usually run into problems when you want to integrate the end result into your (mechanical) CAD software (I use FreeCAD myself). I can refer to the tons of efforts in place and for example issue #186. I think that the most accepted format in CAD is STEP and that it should be supported in KiCAD. Unfortunately KiCAD only supports WRML as 3D format.

In any case, if KiCAD is not going to support (there are not short-term plans to support) other format than WRML, then I think it would be great to actually have **a single place** with all the WRML models and clear guides on how to generate them, so that a person needing a non-available component and wanting to make it (e.g. with FreeCAD) could then contribute it to this single place and be used by others (in my dream, electronic component manufacturers contribute libraries for their products to KiCAD, but that is way ahead). I could even agree in uploading a STEP version and the original "CAD" file towards a future STEP support and allow easy modification of the model. I know this is not a new thing, as I am aware that I am repeating what others have said (and even started doing).

The local generation of files, while it has merit and can be a way to go if we **can not** do better, I think it belongs to the "pre-cloud" era. I could imagine doing this in the 80s in my computer when I had no Internet access. Nowadays, it should be enough "cloud-space" to have everything, complete, in one place. **I want to reuse the work of others and that others reuse mine**.

Now that I have explained where I stand. I regret that there is a constraint in space limiting the inclusion of those 3D models left behind. Yes, I think it would be great to have a **single source** where everybody could use and contribute. **Yes, I can help by contributing part of my time to such an effort**.

Please forgive my rather long post and feel free to "tag it" as "usual EDA libraries rant". In any case **I am very grateful for your models** and for all you have done for the community and for everybody using KiCAD out there.

cbernardo, 2015-10-02:
For a few months now you have the option of creating a STEP model of the
PCB using some scripts + FreeCAD (http://sourceforge.net/projects/kicadstepup)
and there has been some work on an IGES exporter for almost 1 year now

(https://github.com/cbernardo/libIGES). Unfortunately I need to do some
refactoring of the KiCad code before I can introduce the IGES exporter
and at the moment I'm guessing I won't have that ready until at least
the end of this year. IGES has been unmaintained since the last approved
specification in 1996 (and officially abandoned in 2006) but it is still a
widely supported format and much simpler to implement than STEP. So there
will be some proper MCAD support within a year. The work currently being
done with KiCad will also make it possible to provide STEP export via OCE
without the use of FreeCAD, but someone has to write that code. My plan
in the long term would be to provide native STEP support as well but since
I only work on code in my free time I estimate that would be a 2-year task;
the only advantage over support via OCE being that the library would not be
500MB since it will not contain specialized routines which are only required
for interactive mechanical design.

Although you can put as many models as you wish on github I just don't
believe it makes sense. Given the tens of millions of common electronic and
mechanical items used to design and manufacture PCB assemblies, you'd need
very specialized software to help you search a repository and not waste
time downloading too many models. Parts also change with time so you need
to verify the dimensions of any MCAD model you use; with a single site for
all models you can easily waste all your time just searching - it's probably
better to just use google to search for existing models and quickly check
that they are what you want. After all the manufacturer websites have a lot
more information on each model they provide - for the current kicad library
there is no record of provenance for the models.

- Cirilo

imarin2, 2015-10-02:
I see your point. I will have to wait until all manufacturers agree to provide symbol, footprint and 3d model in a standardized way... :)

As for the IGES/STEP support I can only say: **Thank you.**

closed 2015-10-02 by imarin2