KiCad/kicad-library #322: Kicad Github Repo Conventions

SchrodingersGat, 2015-10-26:
Hi everyone,

As someone who is trying to contribute to the kicad schematic/footprint/3d libraries, I have some thoughts on the github repository conventions. Maybe these have already been addressed and perhaps they're inappropriate, but here goes!

1. I believe that the .pretty libraries should be tracked in a single repo as with the schematic symbols. Keeping track of multiple repositories is a pain (and each time I want to pull-request a new footprint in a repo that I don't fork, I need to go through that whole process rather than just making a new branch). It would also be easier for general (non-contributing users) who want to keep up to date with the most up-to-date footprints and only have to do a pull on the master branch of ONE repository.

2. 3d model management - I know that this has been discussed both here and in the kicad forums, and there are mixed opinions on whether they should be with the schematic symbols or the footprints. I personally think that these should be in their own repo - it seems messy having them mixed in with the schematic symbols.

This would keep everything nicely separated and easy to manage. The kicad-library repo works well and covers multiple libraries.

What do others think on this? Managing two logically separate libraries in one repo and then having separate small repositories for the .pretty folders seems disjointed.

CarlPoirier, 2015-11-22:
Hi Oliver,

For number 1, we need it because of the way the KiCad Github plugin works. In the library manager, you indicate the library repository. You can even indicate your own repository, that's what we do to verify libraries before approving the merge request.

For number 2, I would personally like to see the 3D models directly in the .pretties but KiCad developers would need to put lots of efforts into that. Furthermore, it's been discussed many times on the mailing list and it did not seem to lead to some consensus if I remember correctly.

Carl

CarlPoirier, 2015-11-22:
Hi Oliver,

The courtyard clearance for connectors should be 0.5mm as per rule 6.7. Besides that, the silkscreen does not allow one to determine the pin one position once the connector has been soldered on it. That would be great to fix as well. Besides that, it looks great!

timofonic, 2015-12-08:
@CarlPoirier 

Why not change KiCad GitHub plugin to work differently? And even better, make it compatible with other Git providers (even your own Git server).

This change could be extended to make Git usable for storing storage and schematic projects, making collaboration easier.

keruseykaryu, 2015-12-08:
@timofonic: "This change could be extended to make Git usable for storing storage and schematic projects, making collaboration easier."

I have some bad experiences with Altium's CurcuitMaker, and IMHO "clouding" projects is sucks. 
Besides, KiCad has no graphical CVS, which is a minimum to start thinking about that.

closed 2016-08-19 by SchrodingersGat