KiCad/kicad-library #348: A more detailed Kicad Library Convention?

tmanaud, 2015-12-20:
As I'm in the process of creating new footprints (tiny QFN and such) and as I would like to contribute, I try to stick with the KLC.

I've been reading the "PCB Design Optimization Starts in the CAD Library"[1] PDF by PCBLibraries, and it's quite detailed

Sure, the Kicad Library Convention[2] cover few things, but I think if we want more uniform, and standard footprints, more rules have to be added in this document, especially for "new" and small components.

Question: can the KLC be improved, with examples, screenshots, and more cases?

[1] http://media.digikey.com/Resources/Knowles/knowles-sisonic-design-guide.pdf
[2] https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention

f6fvy, 2016-05-16:
A question regarding this doc, especially on section 10.7 (last sentence today):

> 3D Shape ".wrl" files are named the same as their footprint and are placed in a folder named the same as the footprint library replacing the ".pretty" with ".3dshapes".

Is this .3dshapes directory *inside* the .pretty directory, or are they at the same level ?

SchrodingersGat, 2016-05-16:
@f6fvy the 3D shapes directory, confusingly, is located wherever your KISYS3DMOD variable is set. This variable can be set in two places:

1) System variable (this takes priority) 

2) In KiCad (Kicad -> preferences -> configure paths)

Each .pretty directory should have a .3dshapes equivalent as a top-level directory at KISYS3DMOD

e.g. if you have the .pretty dir

> /path/to/Connectors.pretty

then you should also have a path such as

> KISYS3DMOD/Connectors.3dshapes

Does this make sense? I personally think there could be a better way of handling this, but this is the way it's done at the moment.

closed 2016-12-07 by SchrodingersGat