KiCad/kicad-library #354: Introducing generic component-library for certain types of devices?

tahiro86j, 2015-12-30:
As the repetitive process of adding unlisted BJTs to the component-library seems like a never-ending task, introducing some "generic" symbols described as "Type-Footprint-Pinout" (e.g NPN-TO220-BCE, PNP-TO247-BCE) seems like a decent idea.
The very same idea - I imagine - would be applied to some other types of components, too.
Any thoughts?

keruseykaryu, 2015-12-30:
FYI. I have collected so many "repetitive" transistors in `transistors.lib` because I wanted this library was regarded not just as an ordinary source of schematic symbols, but also as a catalog where symbols, footprints, key parameters in fields and PDF links establish a coherent whole.

tahiro86j, 2015-12-31:
Ah, okay. The "transistors.lib" does indeed show traces of your effort, and that should be respected.
Aside from that though, it'd be still useful to leave "independently existing" generic symbols like "NPN-TO220-BCE", "PNP-TO247-BCE" and at the same time, encourage users community to keep adding the PNPs and NPNs not yet listed in "transistors.lib"...how does this sound?
Just saying this because despite the effort you have put into "transistors.lib", those models starting with 2SA, 2SB, 2SC, 2SD (mainly those from Japanese semicon manufacturers) are hardly plentiful in the current status. Of course I'll be doing my best to keep adding as many as I can, but that does not mean all demands would be fulfilled.

keruseykaryu, 2015-12-31:
@tahiro86j: "Ah, okay. The "transistors.lib" does indeed show traces of your effort, and that should be respected."
Considering current situation - I'm not a lib maintainer anymore - I don't care. This was an information "why do so" only.

@tahiro86j: "Just saying this because despite the effort you have put into "transistors.lib", those models starting with 2SA, 2SB, 2SC, 2SD (mainly those from Japanese semicon manufacturers) are hardly plentiful in the current status."
The Japanese ones isn't very popular in Europe, where am I come from. This small set I need to draw schematic for a power amp stage modeled on ROTEL RA-935BX.

tahiro86j, 2016-01-02:
@keruseykaryu First, sorry for the long-response below, and my apologies for any troubles I may be causing to you due to the lack of experience in software development. (I'm an EE student and the this was never the everyday-thing until I decided play with KiCad.)

So, I'm not yet exactly sure of the phrase "coherent whole" you used in the first response to the issue. But what I feared the most now was, there could potentially be multiple symbols with different dimensions defined for different BJTs sharing the same package, pin-layout, and the polarity (NPN,PNP). IMO, this could be quite troublesome because when those using Eeschema attempt to swap a BJT with another type/model, that model simply may not fit into the already-drawn schematic - simply because that model (with the same dimension-physical feature) may have a symbol made by a different person, with a slightly different rule-set, ending up with slightly different dimensions. And I sense that such case is most likely caused by avoiding the use of aliasing across entries of BJTs with the same package, pin-layout, and the polarity (NPN,PNP).

The sole purpose of first opening this issue (well, once closed and then reopened) was so that - at very least - maybe, most BJT-entries to be added in the future could use one of several pre-defined symbols as an alias, peacefully avoiding the above trouble.

Now, I wasn't exactly sure of how aliasing worked when I opened this issue, so I spent the whole afternoon looking at "transistors.dcm" and "transistors.lib" and there I found that aliasing still allows you to specify different descriptions, keywords, datasheet-URL and e.t.c...for each entry defined within "transistors.dcm".
Well, here I was looking at the example of "TIP120" which has;
a.)Two aliases ("TIP121" and  "TIP121") defined in "transistors.lib"
b.)three separate entries for "TIP120", "TIP121" and "TIP122" in "transistors.dcm"

Now, given all above, defining the suggested "generic symbol" (Let's just say in "Footprint_Pinout-Type" this time, for styles like "TO-220_BCE-NPN") now seems like even more of a decent idea.

But even supposing so, I am nowhere near saying "existing entries should also be set to use one of the generic symbols" - no, this is very dangerous and confusing. (The symbol-dimensions should be maintained for already-existing entries, at least for a while.)

Some insightful responses would be quite helpful and very appreciated, since I am now planning on adding to the library, some 30-40 new BJT entries  - those commonly used in Japan and by audiophiles across the globe.

CarlPoirier, 2016-01-02:
Hi Tahiro,

If I understand correctly, you suggest to have all transistors as aliases of generic symbols so that they have more uniform drawings and thus can be more easily swapped in a schematic. Is this right?

tahiro86j, 2016-01-02:
Wow, very nicely summarized!
That indeed is the point!

CarlPoirier, 2016-01-02:
I think it is a great idea but not necessarily all transistors have to work this way at first.

I suggest you start off by creating your generic symbols and have them match the already existing transistors as much as you can. Next, you can create your BJT entries using these. Then, create a pull request and show your work. We'll discuss it further, then.

tahiro86j, 2016-01-02:
@CarlPoirier For existing entries, progress should be made gradually since any quick transition in this kind of situation may cause nothing but confusion and chaos. Thanks for your advice, anyway.

michal777, 2016-01-02:
Some generic transistors symbols are in devices.lib (Q_...)

CarlPoirier, 2016-01-02:
Indeed, but if I am not mistaken we cannot have inter-lib aliases. Putting them in transistors.lib would allow using them. They also seem to nicely match the symbols in transistors.lib, so maybe they could be aliased directly?

tahiro86j, 2016-01-02:
@michal777 @CarlPoirier I shall look into that then!

tahiro86j, 2016-01-02:
@CarlPoirier Alright, #355 should now be ready. Since #355 does not contain any symbols newly added, the real test for me is coming up in coming days when I, as you advised, add those generic symbols and some thirty new entries for Japanese PNPs and NPNs.

tahiro86j, 2016-01-03:
@michal777 @CarlPoirier Alright, made a wrong post here last night. Apologies.
Are there any ways to print the list of symbols within a library file?
I am trying to compare the size of existing symbols for below reasons.
Here, as the base for the generic symbol to be created, I'm thinking of using ones that satisfy;
a.)Symbols for NPN and PNP (Nch, Pch for FETs) are basically identical - except for the position and the angle of arrow.
b.)The outer circle is large enough that it can accommodate the darlington-structure inside.

ricardocrudo, 2016-01-05:
> Are there any ways to print the list of symbols within a library file?

@tahiro86j if you are using linux or mac:

    cat ./library/transistors.lib | grep "DEF " | cut -d' ' -f2

tahiro86j, 2016-01-17:
Well, I was thinking of getting visual images of them, instead of the strings...

CarlPoirier, 2016-01-17:
AFAIK the best you can do is browse for them in KiCad :-(

ricardocrudo, 2016-01-19:
@tahiro86j I wish this feature as well.

tahiro86j, 2016-01-20:
@CarlPoirier As you pointed out, I could browse through symbols just fine, but when trying to compare the actual dimensions of each component, the method you mentioned becomes somewhat useless. Being able to compare them all at once requires symbols be printed on one page (either on screen, or on paper) with unvarying scaling for each entry. The reason I need to do this is described in the 5th post of this thread.

CarlPoirier, 2016-01-31:
Hi @tahiro86j,

The best you can do is place them all on a schematic.

tahiro86j, 2016-02-05:
Ha, why didn't I think of that?

SchrodingersGat, 2017-11-21:
@tahiro86j I am closing this issue due to long period of inactivity. Thanks.

closed 2017-11-21 by SchrodingersGat