KiCad/kicad-library #357: KLC footprint anchor rule problem

g200kg, 2016-01-03:
Hi all,
KLC says footprint anchor should set on pad 1. But I think it had better to allow some exceptions for  mechanical related parts (e.g. potentiometers, switches..).

In these parts, the shaft or button should be placed to accurate position but the pad 1 anchor makes that difficult.For example,

'Button_Switches_ThroughHole' - 'SW_Tactile_SPST_Angled' has pad 1 anchor but the center of the button is not on the grid.

CarlPoirier, 2016-01-16:
Hi,

How is this an issue? Just place your part with an offset of the desired amount. The anchor is set on pad 1 for physical reasons, namely pick and place.

g200kg, 2016-01-17:
Yes, it is possible. But it is so bothering because the correct positions should be calculated part by part mixing mm and inches, also considering part direction.

Though I am not sure which should take priority, this issue or the physical reasons, the pad 1 anchored mechanical parts are cumbersome for me in my experience.

ricardocrudo, 2016-01-19:
@g200kg maybe this could be solved by Graphical Interface, as a feature. Something like yEd software does when moving blocks.

g200kg, 2016-01-20:
I hope so :)
anyway I think that need some mechanical reference point info.

Though I know this is not the best solution, is this KLC compliant?
[image]

DavidMenting, 2016-02-26:
Obeying this rule is very annoying in board layout. It is also very counter-intuitive to have components rotate all over the place, especially when surface mounted devices behave differently. Being able to rotate pots or switches in order to help routing without them shifting with respect to plunger or shaft is very convenient.

I understand this rule helps pick and place, but in the end KiCAD is an application for use by humans, not by machines, right?

My solution now is to duplicate the part, rotate it, drag it on top of the old part and then delete the old part.

anpaza, 2016-05-03:
I think anchor should be just another property of the footprint.
If it's not defined, let's default to (0,0).
But if it is, in the output files that matters the anchor should be used, if it's defined.
This way, the footprint can be defined and used in most convenient way.
Some SMD footprints, for example, aren't symmetric and are defined in drafts wrt some other axes. For example, MicroSD card slot drafts are defined wrt card symmetry line, but card is slightly shifted om the slot.
The anchor could be displayed as a small cross on the footprint.

jkriege2, 2016-10-27:
I also think that for mechanical parts we need some kind of solution that allows to have axes at easily defineable positions ... For the pic and place ... couldn't that be done automatically on export? Just move each component's origin to a pin numbered 1 when exporting?

Best,
JAN

SchrodingersGat, 2017-11-21:
This is now detailed on the KLC pages:

* http://kicad-pcb.org/libraries/klc/F6.2/
* http://kicad-pcb.org/libraries/klc/F7.2/

closed 2017-11-21 by SchrodingersGat