KiCad/kicad-library #365: Which <manufacturer>.lib to choose ?

kinichiro, 2016-01-09:
Hi, I have a question about component library.

- component XYZ was originally manufactured by AAA company
- AAA is acquired by BBB, then component XYZ belongs to BBB, now

In this case, which library should contain XYZ ?
aaa.lib ? bbb.lib ?

I thought if choosing current owner library,
that library will contain many components and grow huge
since acquisitions are so often.

I think big library file that contains many components leads maintenance difficulties.

KLC seems does not have convention about this.

Is there any rule for this issue ?

pointhi, 2016-01-17:
What type of component is it?

Probably it fit better in one of the component specific libraries instead of the vendor specific ones.

kinichiro, 2016-01-18:
Hi,

I am thinking about OTA and BBD chip.
OTA might be contained in opamp group.
BBD might be contained in analog group.
Are these choices correct ?

I asked this question since I can not tell which is preferable library.
Is there any rules for this issue ?

Thanks

pointhi, 2016-01-18:
Hy,

for opamp's ```linear.lib``` and ```analog_devices.lib``` are currently used.

In my opinion ```analog_devices.lib``` should fit for OTA as well as for BBD chips.

keruseykaryu, 2016-01-18:
@pointhi "In my opinion `analog_devices.lib` should fit for OTA as well as for BBD chips."
If they was manufactured by Analog Devices. This is vendor specific library, not function specific library.

pointhi, 2016-01-18:
@keruseykaryu thanks, I have forgotten there is also a company named Analog Devices.

Hm, then I would place the OTA chips into ```linear.lib```. For BBD chips I'm unsure if any current library would fit it.

kinichiro, 2016-01-19:
Thanks.
I thought linear.lib was for Linear Technology,
but it seems for general linear ICs.

kinichiro, 2016-02-09:
Hi,
According to the recommendation, I did 2 things below.

- added OTA to `linear.lib` and submit PR #384 .
- created new `bbd.lib` for BBD chip and submit PR #386 .

I'll close this issue.
Thanks.

closed 2016-02-09 by kinichiro