KiCad/kicad-library #368: JST XH Connectors

BenediktSeidl, 2016-01-09:
I've created footprints and 3d views for JST XH THT connectors: https://github.com/BenediktSeidl/kicad-jst-xh

1. Do you want to include footprints and 3D-Models for JST XH connectors?
2. What do I have to rework? (`check_kicad_mod.py` seems to be satisfied)

Would it be possible to include the scripts? I've already seen that `.wing` files are checked in beside the `.wrl` files. Otherwise it will be pretty time consuming to fix a little bug.

pointhi, 2016-01-16:
kicad_mod gives me a tracepath, but I haven't  figured out yet what causes it. I think some error inside your footprint generator script:

```
Traceback (most recent call last):
  File "./check_kicad_mod.py", line 32, in <module>
    if rule.check():
  File "/home/thomas/Dokumente/Projekte/kicad/kicad-library-utils/pcb/rules/rule6_5.py", line 50, in check
    endComplex=complex(graph['end']['x'],graph['end']['y'])
TypeError: complex() can't take second arg if first is a string
```

There are already JST connectors, made by me ([JST_SH](https://github.com/KiCad/Connectors_JST_SH.pretty), [JST_PH](https://github.com/KiCad/Connectors_JST_PH.pretty)) . Probably we should merge them to 1. Repository, and changing the filenames to something like ```JST_PH_***```. Then you could create a pull-request onto it. I will contact @CarlPoirier to discuss that.

The other way would be adding another new repository, but this could get out of control when adding more and more JST footprints.

About the 3D files, I'm not sure if generated models are allowed yet (without wings3d generating it)

BenediktSeidl, 2016-01-16:
>  I think some error inside your footprint generator script

The problem is that the generated kicad_mod-files contain numbers prefixed with `+`. I could remove them, but kicad seems to allow them, so I've created [pull request #13](https://github.com/KiCad/kicad-library-utils/pull/13) for that. Should have mentioned it, sorry.

Merging all the JST repositories sounds good to me. But I think it would be useful to include the scripts which were used to create the kicad-mod files. Otherwise someone will have to write another script to fix bugs. It took some time to find your scripts for the JST_[SH|PH] connectors.

CarlPoirier, 2016-01-16:
Hi guys,

I also thinkg having one pretty library for all JST connectors would be fine. I'm creating the library.

pointhi, 2016-01-16:
Thx

I think we should think about a generic naming schema of the jst footprins.

I would suggest a prefix like ```JST_PH_02x05_Straight_``` which would help finding the correct footprint while not beeing to long (like the ```Connectors_JST_...```.

For example, a footprint would be named like: ```JST_PH_02x05_Straight_B10B-PH-K.kicad_mod```

I will create the required pull request for SH, and PH in the next days.

BenediktSeidl, 2016-01-17:
Sounds good to me, will also create a pull request for [Connectors_JST](https://github.com/KiCad/Connectors_JST.pretty) in the next days.

[Molex](https://github.com/KiCad/Connectors_Molex.pretty) and the upcoming [Multicomp](https://github.com/KiCad/Connectors_Multicomp.pretty/pull/1) seems to include the whole `Connectors_` prefix so I think we should do the same.

* What about including the scripts to generate the files?
* What about the created 3D files, that are not created by wings3d?

pointhi, 2016-01-17:
Jep, I will change the Multicomp pull request soon, to have more compact names.

Including the scripts would be a good idea, or at least a clear reference where you can find it (In the README.md for example)

The problem is, most people write their own script engines which could result in a big mess of very different scripts.

For example, currently I'm writing a whole new script engine which works similar to a scene tree and should allow very efficient and flexible footprint generators: [pointhi/kicad-footprint-generator](https://github.com/pointhi/kicad-footprint-generator#kicad_mod_tree-next-gen-kicad-module-generator-currently-under-development).

SchrodingersGat, 2016-01-18:
Hi all!

I've been putting together some connector libs too:

https://github.com/SchrodingersGat/Connectors_JST.pretty
https://github.com/SchrodingersGat/Connectors_Hirose.pretty
https://github.com/SchrodingersGat/Connectors_Omnetics.pretty
https://github.com/SchrodingersGat/Connectors_Harwin.pretty

I'll try to incorporate my JST footprints into the new one.

SchrodingersGat, 2016-01-18:
https://github.com/KiCad/Connectors_JST.pretty/pulls

I have made a few pull requests here to start the ball rolling :)

pointhi, 2016-01-18:
Here are my PR:

https://github.com/KiCad/Connectors_JST.pretty/pull/11
https://github.com/KiCad/Connectors_JST.pretty/pull/12

@SchrodingersGat I would recommend you to remove the ```Connectors_``` from the footprint names, because it only extend the name without beein useful.

For example, I named my footprints like ```JST_SH_02x1.00mm_Angled_SM02B-SRSS-TB.kicad_mod``` which contain all important informations on the left side of the name, and the exact footprint name on the right side.

BenediktSeidl, 2016-01-18:
##### `Connectors_` Prefix
[KLC 9.2](https://github.com/KiCad/kicad-library/blob/f41e0370c5dac376f1096984a43f9c70190791e9/KiCad_Library_Convention.txt#L97) says the footprint name starts with the name of the part. I think we should wait for an official statement of the librarian team.

##### Scripts
I really would like to include the scripts directly into the repository as in some years the links could be invalid or at least have to be maintained. An official librarian has to decide and perhaps add a rule to the KLC.

I will hold off on my pull request until all issues are resolved.

SchrodingersGat, 2016-01-18:
Hi @BenediktSeidl  @pointhi  - some good points here and I agree that we need to establish a single naming convention, both for the Connector_ prefix and for the straight/angled/vertical/etc. I agree we should wait for some clarification on this and I am happy to rename all of my parts accordingly.

ricardocrudo, 2016-01-31:
@BenediktSeidl please pull request your scripts to https://github.com/KiCad/kicad-library-utils. Place them in the directory that better fits them functionality. 

As suggested by @pointhi  is preferable to let new footprints without the prefix `Connectors_`.

BTW, the bug fix for support `+` signal is already merged.

CarlPoirier, 2016-01-31:
Hi folks,

There's a catch at removing the Connector_ at the beginning of the footprint name. We won't be able to search for "connector" *by name*. It's true it would be great not to have this repetitive keyword at the start. What I suggest is we start by removing the one in the Molex library and make sure every footprint has the keyword "connector" in its properties instead. We then do the same for the JST and others. Then, we'll be able to search for "connector" *by keyword*. This is when you use the "Add footprints" button in Pcbnew.

As for the orientation, most current footprints use Angled vs. Straight. I say we keep this for a matter of uniformization.

CarlPoirier, 2016-01-31:
As such, the JST pull requests by @pointhi are correct since they have the keyword connector in each of their footprints.

ricardocrudo, 2016-01-31:
@CarlPoirier the new PR's of @pointhi already has tge connector word in the descriptions, so I think is safe remove the prefix. 

@CarlPoirier @pointhi  I'm not sure about have the generic naming for specific part number devices. I mean the `02x1.00mm_Angled`. In this case would be the part number enough? Or maybe move generic info to the end of the name, like this: `JST_SH_SM02B-SRSS-TB_02x1.00mm_Angled.kicad_mod` that's because the part number of device would be more immediately to read.

CarlPoirier, 2016-01-31:
I like the generic info at the end such as in the example you gave: JST_SH_SM02B-SRSS-TB_02x1.00mm_Angled.kicad_mod.

pointhi, 2016-01-31:
@ricardocrudo I proposed the generic info at the beginning to allow easy lookup and logical grouping of the connectors (for example: have all connectors with 15 pins in smd as well as tht in succession)

SchrodingersGat, 2016-02-01:
I've updated my pull requests to match these new naming conventions.

SchrodingersGat, 2016-02-03:
In addition to the conventions above, what about SMT/THRU notation for the connectors? Should we add this as another suffic?

e.g.

JST_SH_SM02B-SRSS-TB_02x1.00mm_Angled_SMT.kicad_mod

pointhi, 2016-02-18:
Ok, here is a proposal for connectors cleanup:

* **Sockets_BNC.pretty** -> **Connectors_TE-Connectivity.pretty** (https://github.com/KiCad/Sockets_BNC.pretty/issues/1)
* **Sockets_WAGO734.pretty** -> **Connectors_WAGO.pretty** (https://github.com/KiCad/Sockets_WAGO734.pretty/issues/1)
* **Sockets_MOLEX_KK-System.pretty** -> **Connectors_Molex.pretty** (https://github.com/KiCad/Sockets_MOLEX_KK-System.pretty/issues/1)
* part of **Terminal_Blocks.pretty** -> **Connectors_WAGO.pretty** (https://github.com/KiCad/Terminal_Blocks.pretty/issues/6)
* part of **Terminal_Blocks.pretty** -> **Connectors_Phoenix.pretty** (https://github.com/KiCad/Terminal_Blocks.pretty/issues/6)

Furthermore, I have done the suggested cleanup for the molex connectors: https://github.com/KiCad/Connectors_Molex.pretty/pull/10

@CarlPoirier @ricardocrudo when you say go, I can do the required rework.

CarlPoirier, 2016-02-19:
@SchrodingersGat Are there SMT and PTH versions of the same connectors?

@pointhi This cleanup sounds reasonable to me. Let's wait for @ricardocrudo's input however. Also one thing you have to keep in mind is we have to keep original repositories untouched so that people using old fp-lib-tables.for-github will not get errors. So the way to go is to create the new library, fill it, adjust the fp-lib-table to add the new pretty and remove the old one, but keep the latter's repository online!

This

SchrodingersGat, 2016-02-19:
@CarlPoirier no generally the part number represents whether a connector is PTH or SMT - however my thoughts were it's nice to be able to tell from looking at the footprint name when you're in CvPcb. Open to suggestion on this though. I've been running with a _PTH and _SMT suffix thus far.

ricardocrudo, 2016-02-19:
@pointhi there is a discussion going on about pretty repos reorganization, please check it https://github.com/KiCad/kicad-library/issues/257 and feed it up with your suggestions (some of them is already there). Michal is also participating with his suggestions.

I gonna start to merge JST connectors.

ricardocrudo, 2016-02-19:
@CarlPoirier I guess the  permissions of KiCad/Connectors_JST.pretty repo need to adjusted, I can't merge on it.

CarlPoirier, 2016-02-20:
@ricardocrudo Fixed. As for the other new repositories, we'll see what we do first with the renaming.

anpaza, 2016-05-03:
So, where are the 3dshapes? I can't find them neither in kicad-library nor Connectors_JST.
I'd say scratch the "uniformity of scripts", give some flexibility to people, everyone uses what it's most used to.

Virtex7, 2017-01-16:
Hi!

i've added your JST 3d shapes, they're working nice.
just wanted to say thanks!

CarlPoirier, 2017-01-21:
All pull requests have been merged, I'm closing the issue.

closed 2017-01-21 by CarlPoirier