KiCad/kicad-library #376: Possibly stale formats in device lib file

freefood89, 2016-01-22:
I was trying to address an issue regarding device.lib, which I assume does not change much anymore. I modified a single component (POT) and saved the lib file to PR. When I tried to partially stage the change (git add -p) I noticed that saving the lib seems to have changed all the components by adding more spaces between coordinate values. I think either the device.lib file in github uses stale formats or Kicad has changed how it formats lib files.

I'm using Kicad 4.0.1 on Windows with git from Command Line.

[image]

nickoe, 2016-01-22:
Hmm, I think this issue is better suited directly on the launchpad bug tracker for the kicad source. You might want to try to see if you can figure out where this change was made and determine if this is a regression or a change by design.

Fat-Zer, 2016-02-23:
The format wasn't really changed... the general format documentation doesn't specify the order of lines inside the DRAW statement neither the number of spaces between fields... But the writer module of the kicad getting improved over time, so it may produce slightly different (but compatable) output like in this case.

In cases like that I usually just resave the library with no changes and make a separate commit with message like "resave the library with recent kicad", or sometimes I redo my changes manually with a text editor. IMHO both ways are ok here.

SchrodingersGat, 2017-10-06:
Closing this issue due to age

closed 2017-10-06 by SchrodingersGat