KiCad/kicad-library #391: How to include base board for shields/modules?

NerdyProjects, 2016-02-11:
Dear all,

I want to use a Core51822 board (http://www.waveshare.com/core51822.htm) in my design.

In my opinion, it would make sense to have a footprint and **not** a template because there could be multiple of this (or other shields) on one board. But anyway, an arduino shield could theoretically also be stacked additionally on something else.
But how would I create an arduino shield with a Core51822? Stacking two templates would not work.

If I create a footprint for that board, that basically is two 2*9 pin 2.00mm pin header with a specific distance and the outline on silk, would that be accepted?

Thanks!

El3ctr0nic, 2016-02-14:
Hi i have same problem with Nucleo64 board. I created development_boards.lib it is in pull reauest queue now. Pin headers aren't good i think. Will be problem with pin numbers. Two pin headers have same numeration and it will be confusing during shematic drawing. Footprint with board dimension silkscreen should be enough. Evryone can draw board outline, and can create various shield dimensions refer to silkscreen.
Soeey for my poor english ;)

Regards,
Marcin

yaqwsx, 2016-02-14:
I have a related question. I switched (or I'm switching) from Eagle to KiCAD. I've been used to special layer in eagle for component footprints. This layer contained drawing of the package. I find it very useful - it was easy to imagine the orientation of the components, PCB modules shape was there etc. However I haven't found any similar layer in KiCAD. Is there? Or is there just a silkscreen and 3D view?

El3ctr0nic, 2016-02-14:
I'm maybe poor expert but its only silkscreen i think. All footpints have
pin numer one in top left corner on default orientation.
 Pin should be marked in some way (check libs and chose some mark way). f.g
in this way. Read KLC for details
[image: Obraz w treści 1]

[image: Obraz w treści 2]

Ten e-mail został wysłany z komputera wolnego od wirusów chronionego przez
Avast.
www.avast.com <https://www.avast.com/sig-email>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

molavy, 2017-10-08:
i need develop custom arduino sheild, i found lots of libs and guide about eagle but none about kicad,
how can i find  arduino shield libs and guides in kicad

poeschlr, 2017-10-09:
Either make a footprint + symbol or make a template project. (The later has the benefit that you have all connectors documented. The former because it is easy.)

molavy, 2017-10-09:
@poeschlr did you answered me?

poeschlr, 2017-10-09:
Yes @molavy  the above is the short version of how this is done. I would advice you ask over at the [forum](https://forum.kicad.info/) if you need a more detailed descriptions.
(This is the repository for the official library. Issues should target the official lib. They are not here for general questions about how to work with kicad.)

SchrodingersGat, 2017-11-21:
Closing, I think this is resolved, @molavy ?

closed 2017-11-21 by SchrodingersGat