KiCad/kicad-library #403: 3D model for a 7mm resistor

Fat-Zer, 2016-02-22:
Currently there is no model for a 7mm resistor in the library. IMHO this is a quite major loss...

If nobody is going to fix it in a couple of days, I'm going to pull-request one, but I will make it with FreeCAD rather than wings.

Any comments / any good ready-to-use models / other volunteers?

Fat-Zer, 2016-02-23:
Ok, I've done a model... I was a bit confused what resistor is suitable for the "Resistors_ThroughHole:Resistor_Horizontal_RM7mm" footprint. I've decided to use dimensions for SFR16S from this datasheet: http://www.vishay.com/docs/28722/sfr16s25.pdf.

[image]

It differs from other resistors, by obvious reasons, but IMHO It looks a bit better.

Shackmeister, 2017-07-01:
@Fat-Zer hi, im case this is not relevant any more could you please consider closing this issue? :)

Fat-Zer, 2017-07-03:
Closing due to outdated as mentioned on https://github.com/KiCad/kicad-library/pull/408.

closed 2017-07-03 by Fat-Zer