KiCad/kicad-library #409: KLC : Inclusion of Fab layer

CarlPoirier, 2016-02-25:
So let's discuss what to do with this layer.

Quote from J-P Louis on the ML: So to summarize, [Fab] layer should have RefDes (Reference Designator) and an outline of the part.
Quote from @michal777 : The rules to convention may be eg: max line width (smaller if necessary), max text size (smaller if necessary), drawing reflects components dimensions, no graphics inside.

I think this is ready to go into the KLC.

michal777, 2016-02-26:
%R can't be plotted if set as invisible regardless of setting "Force plotting of invisible values/references" so shouldn't be hidden (or maybe it's a bug). Then there will be both value and reference on F.Fab and only the value can be disabled (uncheck "Plot footprint values"), %R will always be plotted and it will also be displayed even if References are unchecked (to disable %R all texts or the F.Fab layer must be disabled).
Values are still present on F.Fab. If they can't be plotted without RefDes they can't go to silkscreen (some people may want to print values on PCB), if they will be part of F.Fab as documentation layer they are too big.
Maybe it would be easier if %R were displayed and plotted such as real footprint reference - what about filling a bug report about that (is it really a bug?).

Having components outlines may be helpful for making simplified IDF models (only height is missing).

What about marking a pin 1?

pointhi, 2016-02-26:
> no graphics inside.

To clearify, so something like this is then not allowed:

[image]

CarlPoirier, 2016-10-01:
KLC 1.1: Fab layer contains value, reference designator and an outline of the part.

closed 2016-10-01 by CarlPoirier