KiCad/kicad-library #440: Issue in Cvpcb footprint library path (https://github.com/KiCad)

GauravSupal, 2016-03-10:


Fat-Zer, 2016-03-10:
Against Housings_SOT see this: https://github.com/KiCad/TO_SOT_Packages_SMD.pretty/issues/1

Note that it's likely wrong that they are still referred here e.g. in template/fp-lib-table.for-*

CarlPoirier, 2016-03-12:
Housings_SOT has been marked as deprecated on August 4 and removed on November 15. If you use the Github plugin to have the latest and greatest footprints, it's great, but you need to keep your fp-lib-table up-to-date!! You can get it in the templates folder or use the wizard in Pcbnew. First, clear the table, then in the wizard, select Github repository and just use "https://github.com/KiCad" as the URL. Then select all .pretty repos.

closed 2016-03-12 by CarlPoirier

GauravSupal, 2016-03-14:
Thanks. It is working now. But still some library was missing after reading netlist in PCBnew.