KiCad/kicad-library #443: Wrong pin description for XMega C4

Koot3391, 2016-03-15:
The XMega C4 eschema part has the USART functions labeled for the wrong pins. This is from the current data sheet:
PC0 10 OC0A OC0ALS SDA
PC1 11 OC0B OC0AHS XCK0 SCL
PC2 12 OC0C OC0BLS RXD0
PC3 13 OC0D OC0BHS TXD0

diggit, 2016-03-15:
Confirmed, PD2 is wrong too. Symbols ATXMEGA[16,32]C4-*
@keruseykaryu did you use some script to assign alternate functions to ATXMEGA MCUs?
Commits c323fc3630c0a74f522870bc75596c40ee4a5faa and f77db2ade8bdf240ba7a2060b2dea7c8662e1a14

keruseykaryu, 2016-03-15:
I don't believe in scripts during creating my libraries. So it is my mistake.

diggit, 2016-03-16:
@keruseykaryu can you recheck those symbols and do PR with fix?

keruseykaryu, 2016-03-17:
I'm sorry Patrik, but I have to deny your request. It need to do more than just correct mistakes. Currently, I have neither the time nor the desire for it.

keruseykaryu, 2016-03-20:
Going to the compromise, I can only do this in this way.

Shackmeister, 2017-12-04:
@koot3391 do you know if this has been fixed since? In case not, can you please make a PR for it?

poeschlr, 2018-01-20:
As this repo will be archived soon in preparation for the v5 release, i transferred it over to the new repo https://github.com/KiCad/kicad-symbols/issues/188

closed 2018-01-20 by poeschlr