KiCad/kicad-library #507: Part Library Editor modifies almost all .lib file. Should I create pull request in that case?

alperyazar, 2016-04-24:
Hi,

I am trying to add a new part into "linear" library. **However, my part library editor modifies almost all existing components when I add a new component to this library.**

This is what library editor does when I add a new component called LTC3429:

[linear.diff.txt](https://github.com/KiCad/kicad-library/files/233619/linear.diff.txt)

I don't want to create a pull request for this file because it contains lots of unrelated modifications. Do you have any suggestions?

Application: kicad
Version: 4.0.2-stable release build
wxWidgets: Version 3.0.2 (debug,wchar_t,compiler with C++ ABI 1009,GCC 5.2.0,wx containers,compatible with 2.8)
Platform: Windows 7 (build 7601, Service Pack 1), 64-bit edition, 64 bit, Little endian, wxMSW
Boost version: 1.57.0
         USE_WX_GRAPHICS_CONTEXT=OFF
         USE_WX_OVERLAY=OFF
         KICAD_SCRIPTING=ON
         KICAD_SCRIPTING_MODULES=ON
         KICAD_SCRIPTING_WXPYTHON=ON
         USE_FP_LIB_TABLE=HARD_CODED_ON
         BUILD_GITHUB_PLUGIN=ON

diggit, 2016-04-24:
This is caused by changes in KiCAD library formatting. Use latest KiCAD, or at least build 6331 or later.
It's good to see someone checking diff!

diggit, 2016-04-24:
Latest widows builds: http://downloads.kicad-pcb.org/windows/nightly/

alperyazar, 2016-04-24:
I got the point, thank you.

As you know, my version is the latest stable building and I couldn't find its build number to compare with 6331. I don't know whether this build number is related to commits there: https://launchpad.net/kicad Build 6710 was released on 23-Apr-2016. Build 6331 should have been released more than one years ago. KiCad 4.0.2 was released in February 2016, two months ago. I think, my Kicad should be newer than 6331.

Anyway, is there any base release that every contributer of kicad-library has agreed on it? I expected it to be the latest stable release, not nightly builds. Nightly builds are not considered to be stable, right? How often should I change my Kicad installation if I want to contribute to kicad-library properly?

diggit, 2016-04-24:
Sorry, it was typo. I meant build 6631.
There is no stable release after this revision, so you'll have to stick with nightly. 

No more changes affecting library format were made in later revisions AFAIK, so build>=[6331](http://bazaar.launchpad.net/~kicad-product-committers/kicad/product/revision/6631) is the only condition.

Upstream is quite stable, I build kicad from sources regularly, but it is not needed by for kicad-library contributing.

Even if you use latest kicad, unwanted changes still may occur. KiCAD has no line ordering for some parts of lib defined in standard.

alperyazar, 2016-04-25:
I installed release 6710, now everything is fine. Thank you for support.

closed 2016-04-25 by alperyazar