KiCad/kicad-library #520: KCL Rules 3.1 and 3.2 conflict

inio, 2016-05-07:
> 3.1.  Using a 100mils grid, pin **ends and origin** must lie on grid nodes (IEC-60617).
> 3.2.  *[..]* Large pin numbers can be accomodated by incrementing the width in steps of **50mil**.

Clearly if you increment the pin length by 50mil (for example to 150mil) you can't have both the end and origin on a 100mil grid.

diggit, 2016-05-24:
Only connection **end** of pin matters, it must lie on 100mil grid. **Origin** [0;0] of coordinate system must lie in 100mil grid too.
It simply means, you have to place your pin connection end on [ a*0.1 ; b*0.1 ] coordinates and not sth like [ a*0.1+0.05 ; b*0.1+0.05 ] (still on 100mil grid, but not together with origin [0;0])

At least I understand it this way.

inio, 2016-05-25:
Aaaah, that reading makes so much more sense!

diggit, 2016-05-26:
If it's clear, I think, I can close this issue...

closed 2016-05-26 by diggit