KiCad/kicad-library #529: manufacturer libraries

SchrodingersGat, 2016-05-12:
Is there a guideline that has been developed for determining if components are placed in libraries based on manufacturer or function? Or is this somewhat ad-hoc?

For example, there exist libraries specifically for manufacturers such as Maxim, et al, and there are also function-specific libraries (e.g. interface.lib for components such as RS232 / CAN  transceivers etc). 

As the library grows, this becomes a function of both the suitability of a certain library, and also the size of respective libs. Microcontrollers are a good example of where the parts should be split into manufacturer-specific libs, as there are many many items to choose from and also the functionality is separate enough that people are not going to want to chop-and-change between micros all the time.

I'm interested in what the general consensus (if it even exists) is on this topic.

keruseykaryu, 2016-05-12:
Many years ago, at `kicad-lib-committers` mailing list I proposed a fairly simple division of libraries, manufacturer first, and then the functional group. For example `maxim_switchregul.lib`. Unfortunately, this discussion stood at a standstill.
In my libraries I've long been used this system.

SchrodingersGat, 2016-05-12:
@keruseykaryu I agree with this to a point. I think this has the potential to create a lot of orphaned parts which could be grouped together (in some cases) by function. I guess we'll see what other opinions are floating around out there in the ether.

keruseykaryu, 2016-05-12:
BTW. There is one issue in this system. When one manufacturer has been acquired by other one (last but not least: Atmel by Microchip, Freescale by NXP), what to do? Merge, rename...

SchrodingersGat, 2016-05-12:
I think if we were looking to the future, we could just have a single repo called Microchip_* given their propensity for buying everyone

closed 2016-08-19 by SchrodingersGat