KiCad/kicad-library #536: .lib file descriptions

SchrodingersGat, 2016-05-17:
a) Is there, anywhere, a description of the various symbol libraries and what they should contain

b) Is there a way of storing a library description within the .lib or .dcm file?

SchrodingersGat, 2016-05-17:
The reason that I am asking is that it would be handy to have a proper description of the expected contents of each library, both for users who are looking for a particular part, or who are looking to add their own parts to a library.

Additionally, a description associated with the .lib could be integrated into the KiCad application, so in the library selection window you could see a description of the library.

Perhaps it would be best to have a third file for each library which is simply a description of the contents of the library (plain text or markdown)?

What do people think of this?

wolfgangbremen, 2016-05-18:
This is a thing I am thinking about since a long time. It would be helpful to get deeper information what is in the library. I see a strong relation to issue #529. Ona aim of #529 is to let you recognize what's in it by the **name** of the library, though sometimes the name is too short for that purpose (What is "rfcom" or "silabs" ?) I have to look into the .lib or .dcm-file to get an idea about that.
What about an additional line or section of comment in .dcm (this is where description resides whereas .lib holds graphical information) that decodes the library name? Such that 
# Radio frequency communication (rfcom)
for rfcom.dcm f. e.
Or, if a) more lines or b) proceesing by machine is needed, we mark a section named
$Abstract
  text
$Endabstract
By putting that at the beginning of .dcm, you could simply get an overview over libs by typing
head -4 *.dcm
Unfortunately the .dcm-File is not (yet?) described in the file_formats.pdf.
If our maintainers would agree to an addition like that, I would make a suggestion to extend the file_formats.pdf.

SchrodingersGat, 2016-05-18:
I like the idea of having the .dcm file describe the library contents, except I can see issues with backwards compatibility. Every time someone updates the library, the description will be lost when KiCad re-writes the .lib and .dcm files. Until the KiCad application is updated to preserve library description info, it would be pretty useless, and even after that, anyone who is using an older version of KiCad will delete the description every time they save the library.

A separate file can be maintained independently of the components, and is forwards compatible to eventual integration with the KiCad application code...

SchrodingersGat, 2016-05-18:
e.g. perhaps something along the lines of

    EESchema-LIBDESCRIPTION  Version 1.0
    N Test library
    $DESCRIPTION
    These are the description lines
    multi-lines are allowed!
    $ENDDESCRIPTION
    #End Library Description

closed 2016-12-07 by SchrodingersGat