KiCad/kicad-library #605: 74xx and 74xgxx forced power pins

diggit, 2016-07-06:
All symbols from 74xx and 74xgxx libraries has power pins connected to GND/VCC and set invisible. This is OK in most cases, but I designed several devices with isolated parts and it was problem.
When there are 74* chips on both sides of isolation barrier, you have to modify symbols to stay isolated.

Single part symbols are easy to modify (CNT, REG, MUX,...).
Multi part symbols are problem, there is not space, where you could place power pins (AND, OR, INV, NAND,...).

Does anybody else solved this problem?

SchrodingersGat, 2016-07-07:
I'd really like to see this fixed too. The way I see it there are three options:

a) Leave it as it is, people struggle along with invisible pins.
b) Change all the parts and make the Vcc / Gnd pins visible and give them a non-zero length
c) Create a duplicate of each part with visible pins (as in b) above) so that user have "options"

What do you think we should do @diggit ? I am happy to help with this, I can see it taking some time as there are a lot of parts.

At the same time we should try to improve the general quality of these symbols.

diggit, 2016-07-08:
I'd go b) way, but do you have any idea where to place power pins for multipart symbols like AND,OR,.. ?
- Maybe we can leave power pins where they are and set visible. I am surprised, that same pin name with power symbol is enough to connect it to same called power pin (net). Or is it something else? Is there any documentation about this behavior?

Once this is solved, we can discuss what improvements to do and start working on it.

jkriege2, 2016-10-04:
Hi!

yes, that is a huge pain for me too ... and I've seen this is e.g. also done for the NE555 (even worse). I would suggest a third option though, which I've seen in Target3001!:

d) add an additional part to the component, which contains only the power pins and which can be placed somewhere else on the sheet ... 
[image]

Also this way, we can maybe write a script that removes the existing power pins and adds the new part to the component with defined distances and all ...

What do you think?

Best,
JAN

jkriege2, 2016-10-04:
Eagle also does it wioth option d):
http://dangerousprototypes.com/blog/2013/02/12/eagle-tip-gateswap-and-pinswap-tool/

jkriege2, 2016-10-04:
Just to clarify the problem I made a short test:
[image]

This leads to the following connections:
[image]

i.e. only if pins are invisble, they are automatically connected (555-GND is not tied to the GND net!

Hope that helps,
JAN

diggit, 2016-10-05:
I like **d)** way, but I thought KiCAD is not capable of drawing different graphics for each part of symbol. It is possible! I can't find how to do it in editor, but after direct editing of library file, I was able to do this.

[image]
*I tweaked design too, not important for now, resistor for size comparison*

What do you think about **d)** @SchrodingersGat @pointhi  @CarlPoirier

jkriege2, 2016-10-05:
As far as I saw by now, the only way is this: When drawing a multi-component Symbol, you can choose for each elemnt whether it is visible for all components, or only for the current component. THat's available in the properties Dialog for lines, Pins, ... (shortcut E)

cu
JAN

jkriege2, 2016-10-05:
I can try and make a quick test-comp (e.g. 7400+maybe 7490 to have one multi-comp and one single-comp device) in a branch sometime this week (or on the Weekend) and post the branch-link here ;-)

OK?

cu
JAN

jkriege2, 2016-10-05:
OK,
so I tried to do it for the 74LS00 and 74LS90. Here's the result (as I would like to have it):

[image]

You can look at the lib here:
https://github.com/jkriege2/kicad-library/tree/logic_symbols_with_separate_powercomponent
There is also a test project:
https://github.com/jkriege2/kicad-library/tree/logic_symbols_with_separate_powercomponent/library/doc/test_74xx_separate_power
that demonstrates the parts (then you can easily clone the repo and simply load that .pro-file!)

Doing this I noted several thinks (T), problems (P), KiCAD-problems (KP):
- T1: KiCAD has limited support for different units in one part, i.e. we can draw a different symbol for each unit and have the pins shared or not ... all fine
- T2: The components do not have to be all the same ...
- P1: ... BUT: The REF and Component-NAME are ALWAYS and in all units at the same position. So we either have to make very large parts, so both info fit into the middle (at least with my layout for the power pins), or we need a different layout for power (e.g. as above), or we give up on the requirement that the part is centered (I did it that way for the 74LS90 ... but that's a bit ugly then when moving the parts... all not nice
- P2: KiCAD does not seem to support unit-swapping, right? Otherwise: How can we say that the first 4 can be swapped, but not the last? => IMHO that would mean changes to KiCAD itself, e.g. by allowing us to say all are exchangeable, but Unit E is a Power-Symbol ... and then not exchangeable by default!
- KP1: It's a bit annoying that we cannot set the label-pin-distance for each pin separately, especially since then for clock-pins the text may collide with the clock-Symbol (collision is also for inverted pins and two-digit pin numbers)
- KP2: It would be helpfull, if one could also rotate the pin-label, e.g. for VCC and GND for the power-pin, or at least set the text beside-label/next-to-label property for each label separately. I worked a lot with Target3001! and there you can simply move+rotate the labels freely and while doing so it shows a line connecting the two for reference (like the label/REF on a PCB in KiCAD)
- P3: The whole lib is a mess (really ugly symbols, often not on 100mil grid etc. etc. and don't make me start on KLC
- P4: It would be a hell of a lot of work to go over the whole lib ... does anybody know whether there is a database or something available on the internet that specifies all 74xx/40xx ICs and from which we could generate the lib with minimal manual effort automatically?

Best,
JAN

SchrodingersGat, 2016-10-05:
Jan,

Regarding `P4`, I have considered this before, and perhaps something like [this](https://en.wikipedia.org/wiki/List_of_7400_series_integrated_circuits) would be a good starting point?

SchrodingersGat, 2016-10-05:
@diggit further to what @jkriege2  has said, here's how you specify which sub-part a graphics item is related to:

[image]

jkriege2, 2016-10-06:
Hhmmmm (about WP-page): Yes that could be a simple start. I searched a bit yesterday, but didn't find anything that could be converted automatedly. So I was thinking about starting with a script for the single gates, as they usually have a default layout (pin-compatibility!!!). The script should do the following:

- Say we want to alter the symbol for 7400, 74HC00, ... (i.e. first input a list of aliases)
- Then search for the record in the lib and replace it by the new record. The new record will be created by a simplified dataset as follows:
    1. name + aliases
    2. description+keywords+PDF-link (can maybe be generated automatically from WP & Texas Instruments ...)
    3. list of gates, e.g. ["NAND2", "NAND2", "NAND2", NAND2"]
    4. pins, either default-layout, e.g. when giving 4 2-input-gates, or giving 6 1-input-gates, or list of gates in a layout like this [I1.1, I1.2, O1.1, I2.1,I2.2,O2.1,...] ->[1,2,3,4,5,6,9,10,8,12,13,11]
    5. VCC+GND pins
- Then the script has hard-coded instructions of how to code the standard-gates (+power symbols) and we can at least generate update the gates.

For the other devices we would need more effort, either do them by hand, or do something like this: Have a file with entries in the format `# PIN NAME TYPE [LRBT]` for the single pins. Then one could have special record --- for a horizontal line and say === for typical separation of the indention form (see e.g. my 7490 between top and bottom half), e.g.:

> ? UNIT
> 2 R0.1 I L
> 3 R0.2 I L
> 6 R9.1 I L
> 7 R9.2 I L
> ===
> 14 ~CP0 CI L
> 12 Q0 O R
> ---
> 1 ~CP1..3
> 9 Q1 O R
> 8 Q2 O R
> 11 Q3 O R
> ? PWR
> 5 VCC
> 10 GND

I used `?UNIT` to start a new subunit and `?PWR` to start the power symbol. I'm not yet clear, how one could may the inside `&`-gates ... either we leave them out, ior alter them by hand later, or something like that (the & is a free TEXT that will be place within the innner compartment):
> ? UNIT
> ~~~ &
> 2 R0.1 I L
> 3 R0.2 I L
> ~~~
> ~~~ &
> 6 R9.1 I L
> 7 R9.2 I L
> ~~~ 
> ===
> ...
The `~~~ TEXT` would start a left inner box with all pins in it, until the next `~~~`. Then one would maybe have `+++` for an inner box from the right...

This way we could at least easily define the parts by hand once and the regenerate the library by only changing the script (i.e. the script defines the style). Also the script would not delete parts that were drawn by hand and for which there is no definition available in the script- ... does this seem good to you? But ... there are still a lot of things to clear up, e.g. where to store the script etc.

Best,
JAN

davidsrsb, 2016-11-23:
Why do we waste effort on the DeMorgan equivalents? I have never seen them used outside a lecture theatre.
I would be more interested in offering the octal parts as both full block and 8 sub-modules

lichau, 2016-11-23:
Jan, thanks for pointing me here.  My previous experience is with option d)--a separate symbol for power and ground.  While I understand the reason for hidden pins, I personally don't like them.  I screw up enough without stuff happening that I didn't do.  Or, do explicitly.
 I typically put the power/gnd blocks on a separate schematic sheet--usually with associated bypass capacitors.  That unclutters the main schematic sheet.

For large parts with many Vcc/GND pins, it seems that either hidden pins or a separate block are about the only way to go.  

One of my reasons for considering KiCad is the schematic level error checking.  I really don't mind getting  a bunch of error flags, as long as I can turn them off.  In fact, I like it--makes me affirmatively turn them off.

Thanks for the help.

Rich, the NOOBie

koder68, 2017-01-14:
I was missing some circuits in the 74xx library and wrote a request to update it but now when I have read through this thread I realize things are moving in another direction. This is my opinion:

While I like the hidden power pins alternative d) is very reasonable. But going for a generationation strategy there is no need to choose. And the alternative symbols of library 74xx and ttl_ieee can be done at the same time.
If we create a file that describe the circuits functions it should be possible to create generators that create the libraries we want. One with the kind of symbols used by most parts in 74xx library. An alternative with separate power components, and the libraries with the more complex symbols of ttl_ieee can be generated from the same source file in variants with separate power pins and with hidden ones.

The tricky will be to create the description file so that it does not contain any presentation details while having enough information about function to generate the libraries. And of cource the de-morgan variants should stay.

I expect that the same strategy an be used for at least the 4000 library and likely for other components as well.

In a future the generation of the separate libraries should be replaced with direct generation within the schematic editor. Perhaps I would like to have unit A of a 74HC00 shown as the "old" style and unit B as the "new" style.

Regarding file formats I highly dislike making up new "flat file" formats as described above. An XML based format should be used to enable use of lots of premade code for specifying, parsing and checking.

jkriege2, 2017-01-14:
Hi!

please make a proposal for a fileformat then ... generally using XML is a god idea from the viewpoint of the tool (you don't need to implement the parser ...), but especially XML (but also JSON) has a VERY large overhead (especially when few info is stored, thing e.g. of a tag:
> <pin id="15" name="Q1" type="O" oreintation="R"/>
instead of a simple line:
> 15 Q1 O R
and is really pittysome to write by hand ... therefore I proposed such a kind of flat format!

Best,
JAN

bobc, 2017-06-21:
I thought I would have a go at writing a generator script as suggested by @jkriege2 

For an input file

```
#
COMP 74xx574 IC
UNIT
2 D1 I L
3 D2 I L
4 D3 I L
5 D4 I L
6 D5 I L
7 D6 I L
8 D7 I L
9 D8 I L
SPC L
11 CLK CI L
1 ~OC I L
19 Q1 O R
18 Q2 O R
17 Q3 O R
16 Q4 O R
15 Q5 O R
14 Q6 O R
13 Q7 O R
12 Q8 O R
END
UNIT PWR
20 VCC P T
10 GND P B  
END
#
COMP 74xx595N IC
UNIT
14 SER I L
SPC
11 SCK CI
10 ~SCL I
SPC
12 RCK CI
SPC
SPC
SPC
13 ~G
15 QA I R
1 QB
2 QC
3 QD
4 QE
5 QF
6 QG
7 QH
SPC
9 QH*
END
UNIT PWR
16 VCC P T
8 GND P B  
END
````
The pin type and direction will default to the previous definition, or Input, Left if none. Pins are assigned in a top down fashion independently for left and right in the order they appear in the file.

The script generates a new library (for now) containing the components.

These look like this , Unit A
[image]

Unit B
[image]

This only handles simple box elements, and there are things like origin point to tidy up, etc. I will put the scripts in a github repo later.

bobc, 2017-06-23:
I've added code to handle NOR/OR, NAND/AND and a few other changes. The code is at https://github.com/bobc/kicad-symgen
Sample output
[image]

poeschlr, 2017-06-23:
Very nice.
Is it possible to fill the gate shapes as well?

bobc, 2017-06-23:
Sure, no problem. With 10 mil border?

poeschlr, 2017-06-23:
Might be a good idea. (I have not noticed that the border is thinner then the other units.)

bobc, 2017-06-23:
> Is it possible to fill the gate shapes as well?

Actually, maybe not so easy. OR gates do not fill properly
[image]
I would have to replace the arcs with a polyline.

jkriege2, 2017-06-26:
possibly one has to put a filled polygon BEHIND the lhs of the gate ... 

In genertal: Exactly what I proposed :-))) GREAT WORK!

As you're currently designing the gates: I think in general they should be significantly smaller ... I tried to get that done within the KLC (see my screenshots above, post from Oct 5th) ... maybe we could even make a KLC exception for gates that it is OK to place pins on a 50mil raster (as we do for small devices, such as R,C,diodes,transistors, ...) @SchrodingersGat What do you think on a KLC-exceptio for that?
Best,
JAN

bobc, 2017-06-26:
I couldn't find any way to fill OR gates without using a polyline to trace the complete outline. I am reluctant to do that, because I don't want to add extra drawing overhead to eeschema, which can be quite slow on moving large chunks of schematic.

It's very difficult to make the gates smaller and keep to KLC, so a waiver would be very useful. 74xGxx lib has small symbols, which for consistency would be nice to re-use, but the 74xGxx symbols fail KLC check in many ways.

I have a logic symbol set that is about 70% smaller than the current symbols, which is KLC compliant but probably still not as small as some would like.

[image]

diggit, 2017-07-04:
Here is my entry to gate design contest!
Pins are on 100mil grid. Pins could be smaller without numbers. Sorry for noodle aspect ratio. Ignore pin numbers and pin types.
[image]
Some passive components for scale.

evanshultz, 2017-07-04:
Personally, a dislike having a box for the power pins. I prefer them to be two floating pins. Yay or nay?

diggit, 2017-07-04:
Pins with intelligent spacing (distance like capacitor or so) should be OK. I am ok with or without box.

bobc, 2017-07-05:
Well, someone objected to the "floating pin" format, so I changed to box. If you like, I can create an alternative (de morgan) which has no box. 

(It's a pity there is not more than two alternatives, otherwise we could create a complete set of alternative symbols and the user could scroll through until they find one they like).

@diggit Those look great, I will try to adopt them. Even better if you could post a .lib file with those shapes in :)

SchrodingersGat, 2017-07-05:
> (It's a pity there is not more than two alternatives, otherwise we could create a complete set of alternative symbols and the user could scroll through until they find one they like).

It's a good start, and IIRC the "new" lib format allows for `N >= 1` symbol display versions.

diggit, 2017-07-05:
@bobc sure, here it is [gates.zip](https://github.com/KiCad/kicad-library/files/1126017/gates.zip), it was just concept, so it is all one symbol. Most of time, I was tweaking OR gate size and pin lengths. Pin and lines are there on purpose to align pin numbers properly for normal and inverted outputs...

PR #1578 (2017-09-21): 74xx rework (remove hidden power pins)
## Purpose

The primary goal is to resolve #605. Because the fix for hidden power pins require many incompatible changes, an opportunity is taken to rework the library to be much closer to current KLC, as well as create a more compact set of symbols.

## Summary of changes
- remove all hidden power pins
  - power pins on unit for single units
  - power pins in separate block if multiple units
- uniform width where possible (600 mil)
- smaller logic symbols (300x300)
- shorter pin length (150 mil)
- fixed several errors in pinouts
- compliance with KLC
  - [... cut]
closed 2017-09-21 by SchrodingersGat