KiCad/kicad-library #633: Helping the library maintainers

yaqwsx, 2016-07-30:
Hello,

when KiCAD 4 came out and brought centralised library, which is community-developed, I immediately switched from Eagle to KiCAD, as the libraries are the biggest pain in PCB CADs and I think community-developed library is the right way to go (at least for open source project).

I watch the library and I also tried to contribute multiple times. I think the contribution process is really, really slow and it degrades the wonderful idea of community based library. There is also a ton of months old PRs with no reply! I think all of this may be harmful to the whole project because it can discourage people from further contributions and also leads to duplicates:

- many opened PR are mess, people do a lot of duplicate work (because they didn't noticed already existing PR) and it can discourage potential contributors

- the changes are merged slowly, the upgrade process of local library in KiCAD is unclear. I think almost every user wants fast deliver of the library!

- when there is no response to the PR, contributor might feel unwanted and will never contribute again...

Don't get me wrong, I don't want to complain or blame you. I understand that there is a lot of work with such a huge library maintenance and there are not enough people and time. I just want to point the obvious problem and ask (and discuss):  

## What can we (the contributors) do to help you (maintainers)?

I really like the library idea. I want to help it happen (and I think I am not alone who wants to help). What should I do (when contributing or when I have a free time I want to invest in KiCAD) to help you the most?

## My notes

- I noticed there are screenshots of footprints/symbols attached to some of the PR. I find it useful, does it help you, when reviewing the PR? Or is it useless (and I should not do it)?

- Do you like having PR commented/reviewed by the community? Many maintainers hate it my experience (they probably think somebody tries to to their job) and it discourages me from PR commenting (like "This PR seems OK to me and if I was the maintainer I would merge it", "This PR seems dead to me, shouldn't it be closed", etc)

PS: I am not sure if issue is the right place for asking this/opening the question, however I haven't found a better alternative (enough readers, permanent/history, etc)

SchrodingersGat, 2016-07-31:
Thanks for the input :)

**Screenshots**
Where appropriate, screenshots can be *very* helpful, especially to demonstrate a simple change.

**Adding notes**
Many PR are submitted without any description (or without a *good* description). A succinct note describing what you have added or changed is an amazing help and assists in directing the focus of the librarians.

**Commenting on open PRs**
I don't think any librarian will complain that someone is trying to assist - there's a lot of open PRs and only few librarians. As long as the comments are helpful or drawing attention to an old PR for example, I don't see a problem. Any long tangential discussions should take place elsewhere. If someone is commenting too much, one of us will certainly mention it.  

**Know the KLC**
The biggest help is knowing good practice for submitting PRs that fall within the library convention.

* Look at the KLC rules 
* Look at other similar parts / footprints
* Be consistent as far as reasonable 
* Ask if unsure 

I hope this helps. Anything that makes the lives of both the contributors and the maintainers is only a good thing.

jacobrosenthal, 2016-08-19:
We could use [github templates](https://github.com/blog/2111-issue-and-pull-request-templates) to list your bullet points to check. But for something like screenshots instead of forcing the user, lets automate that with a bot that can screenshot and run whatever python scripts are already run against them.

SchrodingersGat, 2016-08-19:
The github template looks like a cool feature - was not aware of that. I agree that the screenshots should not be the work of the user, it's something that should be automated.

jacobrosenthal, 2016-08-19:
Does anyone have a good idea on grabbing a screenshot programmatically? I saw cpavlina might have done something in https://github.com/cpavlina/kicad-schlib-preview/issues/1

What scripts should be run against the incoming libraries?

SchrodingersGat, 2016-08-19:
The work that @cpavlina has done there looks great! (I wish I had known about this earlier as I started to write such a script myself! There's a few hours I'll never get back)

This looks like it would be the tool to use. An interesting idea would be to add a graphical-diff feature, highlighting changes between the images!

KLC scripts (as they currently stand) are here https://github.com/KiCad/kicad-library-utils - they have a description with them also

SchrodingersGat, 2016-08-19:
If we can find / make / co-opt a library to make image previews of the symbols and the footprints, this should be included into kicad-library-utils.

jacobrosenthal, 2016-08-19:
@SchrodingersGat Response from cpavlina 

> https://github.com/cpavlina/kicad-schlib/blob/master/Makefile
https://github.com/cpavlina/kicad-schlib/blob/master/scripts/schlib-render.py

Ill take a look at that eventually if you don't, but I'm still taking a look at ci options and how the hell to tell which files have changed. Were an odd case here because were not testing the entirety of the code base like normal ci would, I just want to know which files were touched in all commits to be merged, and run those python scripts on them..

SchrodingersGat, 2016-08-19:
It's even worse than that - for the case of the .lib and .dcm files, we only want to look at the individual components that have changed (within those files!)

The .pretty repositories are easier as the footprints are stores as individual files.

ricardocrudo, 2016-08-19:
> Does anyone have a good idea on grabbing a screenshot programmatically? I saw cpavlina might have done something in cpavlina/kicad-schlib-preview#1
> 
> What scripts should be run against the incoming libraries?

Yes, I did. Check it here: https://lists.launchpad.net/kicad-developers/msg23836.html

Unfortunately, I didn't have time to go ahead with the development and make a proper implementation. Ideally, a tool like this would be integrate with the, already working, checking scripts on the Jenkins. 

Another idea I have, and I'm sure it is very helpful, is to attach the pins information (e.g.: the pin type) along with the PR image or (probably easier) in the PR comments. Like following.

[image]

In general, make more information available in the PR, surely will make the evaluation process faster. By "make more information available" I mean automatically. However, while such integration is not working yet, I agree with @jacobrosenthal, a template might help.

jacobrosenthal, 2016-08-19:
I see the current tasks as
- [ ]  Get the files affected via the PR object or via the github pull api https://developer.github.com/v3/pulls/ so also need to find a ci provider that actually passes through the github pull request json from the web hook (drone.io doesn't it seems..)
- [ ]  Find out if either of the existing screenshot scripts will work headless on a server or docker container
- [ ]  Figure out how to isolate the .lib and .dcm components affected by the change

jacobrosenthal, 2016-08-20:
Pretty first because thats way easier. Was hoping to do this as a one liner but python script was a pain to alter to take std in file list, are we up for altering that? Also it should be altered to give an exit code.
```
#!/usr/bin/env bash
FILES="$(curl https://api.github.com/repos/KiCad/Shielding_Cabinets.pretty/pulls/6/files | jq -r '.[] .filename')"
python check_kicad_mod.py $FILES
```

jacobrosenthal, 2016-08-20:
blah, forget argparse
```
./check_kicad_mod.py `curl -s https://api.github.com/repos/KiCad/Shielding_Cabinets.pretty/pulls/6/files | jq -r '.[] .filename'`
```

SchrodingersGat, 2016-08-21:
Sweet :) I'm making some good progress on integrating script-based rendering into the KLC scripts also!

jacobrosenthal, 2016-08-21:
@SchrodingersGat you wanna add exit codes to check_kicad_mod.py. Im presuming any violation at all is a non zero exit?

jacobrosenthal, 2016-08-21:
Seems like no reason not to test the entired changed library files, even if they didnt touch all of them. If anything the unrelated violations could be ignored or better yet, we can ask the submitter to clean that file up for us?
A similar one liner that just throws dcm and lib at checklib.py dcm seems to just bounce off it so probably ok.
```./checklib.py `curl -s https://api.github.com/repos/KiCad/kicad-library/pulls/616/files | jq -r '.[] .filename'````

Which file is supposed to check dcm? 

Im working on figure out the line number changes so we can only make screenshots of the appropriate components still.

jacobrosenthal, 2016-08-21:
I cant find a good rule of thumb for how many context lines are included in order to remove them from the result, might have to just count them (best way would seem to be lines that start with a space

Heres a one liner for getting line numbers from the patch...Still has the comma in it, but i guess well split on that in the next step
```curl -s https://api.github.com/repos/KiCad/kicad-library/pulls/616/files | jq -r ".[]  | select(.filename == \"library/dc-dc.dcm\") | .patch" | grep -Po '\+\K[0-9]+\,[0-9]+'```
```
1,11
208,12
670,12
784,64
854,47
1021,12
1063,12
1098,16
1306,4
```

SchrodingersGat, 2016-08-22:
My *nix-fu skills are somewhat lacking, so I might not be much help with your scripts above. 

However I did have an idea for working out which components have been changed within a library.

We already have a set of python scripts that can parse a .lib file and extract symbols. If we add a line to the Component object that calculates a simple checksum, then you can run the old file and the new file through the script, and only throw up output messages for components where the checksums have changed, or if they are new components?

Would this be easier or harder than looking at the raw file information? I'm not sure how the script would access the "master" version of the file (does it pull it down from github?)

jacobrosenthal, 2016-08-22:
Sounds simpler... except for getting the original file.. Most ci will pull and merge the pr, but yeah git clone or something

jacobrosenthal, 2016-08-23:
clone a bit heavy, looks like you can wget or curl https://github.com/KiCad/kicad-library/archive/master.zip

jacobrosenthal, 2016-08-25:
I note the Kicad project uses Jenkins, but Im not familiar and thats pretty heavy to get started on so I just put it up on circleciI for no. Heres what what Ive posted so far for kicad_mod's 
so you can see what Commit Status API result looks like:
https://github.com/jacobrosenthal/Shielding_Cabinets.pretty/pull/1 

You can PR into my fork to see the results there or just grab the test.sh script out of master and run it locally, you just need to provide it the environment variables the ci server would be giving it like
```CIRCLE_PROJECT_USERNAME="jacobrosenthal" CIRCLE_PROJECT_REPONAME="Shielding_Cabinets.pretty" CI_PULL_REQUEST="https://github.com/jacobrosenthal/Shielding_Cabinets.pretty/pull/2" ./test.sh```

CarlPoirier, 2016-08-28:
Hi @jacobrosenthal,

Your input on the matter is greatly appreciated!

I do like the part checksum idea. Is it possible to store somewhere the checksums of the master branch and simply update them when needed instead of downloading the whole repo?

SchrodingersGat, 2016-08-29:
Have a look at how I've implemented the checksum here - https://github.com/KiCad/kicad-library-utils/issues/47

You only need to look at the .lib files that have actually changed, and the checksum doesn't need to be stored anywhere

SchrodingersGat, 2016-08-29:
Sorry, here's the actual code. https://github.com/KiCad/kicad-library-utils/pull/48

CarlPoirier, 2016-08-31:
And it's merged!

jacobrosenthal, 2016-09-01:
Since we already have a nice list of files, we can just wget them individually from raw.githubusercontent.com rather than the entire kicad-library since its so big

Anyone have a preference on screenshot methodology?

Who needs help with what.

jacobrosenthal, 2016-09-01:
Seems like youll want to probably do more of this in python yet, but Ill still keep problem solving in bash

Playing with this script https://github.com/jacobrosenthal/kicad-library/blob/master/test.sh to wget changed files and throw them at the new comparelibs.py

I posted here https://github.com/KiCad/kicad-library-utils/pull/48#issuecomment-243970675 but I think we could make comparelibs.py more parseable for the next step with + and - signs for components changed.

SchrodingersGat, 2017-10-06:
Closing this issue due to age.

closed 2017-10-06 by SchrodingersGat